ROBOTIS-GIT/turtlebot3 #357: Picam freeze

arixrobotics, 2018-12-16:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [/] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [/] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [/] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [/] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x] not sure, downloaded in December 2018
    - Firmware version: [x.x.x] not sure, using the one in the downloaded Raspbian image
 
6. Write down the commands you used in order
    on laptop:
    - roscore
    - rqt_image_view

    on tb3:
    - roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch 

7. Copy and Paste your error message on terminal

> pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch 
> ... logging to /home/pi/.ros/log/13f84b2a-00f2-11e9-a504-6c71d97cbaff/roslaunch-raspberrypi-1525.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://192.168.1.218:39943/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /raspicam_node/camera_frame_id: camera
>  * /raspicam_node/camera_info_url: package://turtleb...
>  * /raspicam_node/framerate: 15
>  * /raspicam_node/height: 480
>  * /raspicam_node/width: 640
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.13
> 
> NODES
>   /
>     raspicam_node (raspicam_node/raspicam_node)
> 
> ROS_MASTER_URI=http://192.168.1.57:11311
> 
> process[raspicam_node-1]: started with pid [1534]
> [ INFO] [1544938851.012345956]: splitter component done
> 
> [ INFO] [1544938851.193950142]: camera calibration URL: package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml
> [ INFO] [1544938851.261482438]: Camera successfully calibrated from default file
> [ INFO] [1544938851.261671786]: using default calibration URL
> [ INFO] [1544938851.261839100]: camera calibration URL: file:///home/pi/.ros/camera_info/camera.yaml
> [ INFO] [1544938851.262271917]: Unable to open camera calibration file [/home/pi/.ros/camera_info/camera.yaml]
> [ WARN] [1544938851.262382036]: Camera calibration file /home/pi/.ros/camera_info/camera.yaml not found.
> [ INFO] [1544938851.262492154]: No device specifc calibration found
> [ INFO] [1544938855.803032683]: Starting video capture (640, 480, 80, 15)
> 
> [ INFO] [1544938855.804324729]: Video capture started
> 
> 
> 

8. Please, describe detailedly what difficulty you are in 

    - when I turn on my pi camera, it works for only a short amount of time (less than 1 minute). 
    - after that time, no new image is published.
    - but there is no error printed
    - if I restart the publisher node (turtlebot3_rpicamera.launch), it works OK again. 
    - but, after some short time, it stops working as before.

routiful, 2018-12-19:
Have you tried below instructions?
http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/#appendix-raspberry-pi-camera

WLwind, 2019-01-07:
Have you solved this problem? It comes to me too. The raspicam always works for less than 20s, and then stops publishing images.

arixrobotics, 2019-01-21:
@WLwind I haven't solved this problem. Haven't got the time to revisit it. If you managed to find a solution, please post here too.

kijongGil, 2019-01-23:
@WLwind, @arixrobotics
This issue occurred after updating raspicam_node package.
One way of solution is that remove this package and reinstall old version.

arixrobotics, 2019-01-23:
In the raspbian image provided by Robotis, the package raspicam_node is already provided, correct? So we should not git clone the repo as per the instructions here http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/#appendix-raspberry-pi-camera ?

arixrobotics, 2019-01-28:
Ok I can confirm that using the pre-installed raspicam_node package in the provided image worked and do not freeze. 
Please update the emanual so that other users do not try to install / update theirs. 
Thank you!

closed 2019-01-28 by arixrobotics

rohbotics, 2019-01-28:
Hi, maintainer of raspicam_node here.

We were tracking this issue here https://github.com/UbiquityRobotics/raspicam_node/issues/65, and it should be resolved on the master branch now.

In the future don't hesitate to ping me for pi cam related issues, this bug could have been squashed a lot sooner if I knew about this thread.

Rohan
Ubiquity Robotics

kijongGil, 2019-01-28:
Thank @rohbotics :) 
If we have an issue with pi cam, we will do it.

Gilbert.