ROBOTIS-GIT/turtlebot3 #361: Unable to open Serial Monitor

anwaybose, 2018-12-21:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ 1] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [1 ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [1 ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [1 ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    Hello,
I am using turtlebot3 waffle-pi to make a 4 wheel drive mecanum robot. I have made some changes to receive encoder data in ROS.
But now whenever I upload the code it is getting uploaded. But then no port is showing in tools -> port, and I am unable to open the serial monitor. Every time I need to enter the Firmware Recovery mode to upload any code after that. for your convenience I'm attaching my codes here.
[turtlebot3_mecanum.tar.gz](https://github.com/ROBOTIS-GIT/turtlebot3/files/2701987/turtlebot3_mecanum.tar.gz)

routiful, 2018-12-22:
Did you check your Dynamixel ID and baudrate?
You should check line 47~ 51 In turtlebot3_mecanum_motor_driver.h

anwaybose, 2018-12-22:
@routiful ,
I have checked them, they are all fine. Due to some bug my code is corrupting the firmware everytime I upload it to the OpenCR.

routiful, 2019-01-02:
You can use debug serial port to find where the error is occurred

closed 2019-01-25 by kijongGil