ROBOTIS-GIT/turtlebot3 #362: multi-turtlebot3 navigation in real environment

wx0205, 2018-12-22:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [√ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [√ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [√ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [√ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    Software version: [1.12.14]
    Firmware version: [1.2.3]
 
6. Write down the commands you used in order

    - roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml
 
7. Copy and Paste your error message on terminal
When I run turtlebot3 navigation, the output like this:
    - [ INFO] [1532580173.243852203]: Got new plan
[ WARN] [1532580173.245276133]: DWA planner failed to produce path.
[ INFO] [1532580173.443843662]: Got new plan
[ WARN] [1532580173.445304680]: DWA planner failed to produce path.
[ WARN] [1532580173.643893361]: Clearing costmap to unstuck robot (3.000000m).
[ INFO] [1532580174.043834647]: Got new plan
[ WARN] [1532580174.045632208]: DWA planner failed to produce path.
[ WARN] [1532580174.243951804]: Rotate recovery behavior started.
[ERROR] [1532580174.244415898]: Rotate recovery can't rotate in place because there is a potential collision. Cost: -1.00
[ INFO] [1532580174.643841480]: Got new plan
[ WARN] [1532580174.645396002]: DWA planner failed to produce path.
[ WARN] [1532580174.843851086]: Clearing costmap to unstuck robot (1.840000m).
[ INFO] [1532580175.243841053]: Got new plan
[ WARN] [1532580175.245196120]: DWA planner failed to produce path.
[ WARN] [1532580175.443900326]: Rotate recovery behavior started.
[ERROR] [1532580175.444206757]: Rotate recovery can't rotate in place because there is a potential collision. Cost: -1.00
[ INFO] [1532580175.843830584]: Got new plan
[ WARN] [1532580175.845862800]: DWA planner failed to produce path.
[ERROR] [1532580176.043937473]: Aborting because a valid control could not be found. Even after executing all recovery behaviors

  
8. Please, describe detailedly what difficulty you are in 
   When I run navigation with turtlebot3  in real environment. In rviz, it can get a path, however, the turtlebot3 don't move, and the output in terminal like this. How can I solve this?
   Thanks!

wx0205, 2019-01-02:
Thank you very mach! I got that when I had trid it many times.
However, when I run the package of multi-turtlebot3 navigation which I wrote. 
I do like this:
[PC] roscore
[turtlebot3]  roslaunch turtlebot3_bringup multi_robot.launch 
[PC] roslaunch navigation_wx navigation_multi.launch map_file:=$HOME/map.yaml

I got tf, there are /tb3_2/base_footprint->/tb3_2/base_link,  /tb3_3/base_footprint->/tb3_3/base_link, /tb3_4/base_footprint->/tb3_4/base_link,  odom->footprint

I think there are odom->footprint for each turtlebot3, however, I can't get it. I send goals for each turtlebot3 and want them navigate Independently in real environment. 
Can you give me some advise? Thank you!

navigation_multi.launch:
```
<launch>
  <!-- Arguments -->
  <arg name="model" default="waffle_pi" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="open_rviz" default="true"/>
  <arg name="map_file" default="$(find navigation_wx)/maps/map.yaml"/>
  <arg name="robot_1"  default="tb3_2" />
  <arg name="robot_2"  default="tb3_3" />
  <arg name="robot_3"  default="tb3_4" />
 
  <param name="/use_sim_time" value="false"/>

  <!-- Map server -->
  <node name="map_server" pkg="map_server" type="map_server" args="$(arg map_file)">
     <param name="frame_id" value="/map" /> 
  </node>

  <!--  *******first turtlebot3: named tb3_2*********  -->
  <group ns="$(arg robot_1)">
   <param name="tf_prefix" value="$(arg robot_1)" />

   <!-- Map server -->
    <node name="map_server" pkg="map_server" type="map_server" args="$(arg map_file)">
       <param name="frame_id" value="map" /> 
   </node>

   <!-- Turtlebot3 --> 
   <include file="$(find navigation_wx)/launch/robot_model.launch">
      <arg name="model" value="$(arg model)" />
      <!-- arg name="robot_name" value="$(arg robot_1)" --> 
   </include>

   <!-- AMCL -->
   <include file="$(find navigation_wx)/launch/multi_amcl.launch">
      <arg name="robot_name" value="S(arg robot_1)" />
   </include>

   <!-- move_base -->
   <include file="$(find navigation_wx)/launch/move_base.launch">
      <arg name="model" value="$(arg model)" />
   </include>
  </group>

  <!--  *******second turtlebot3: named tb3_3*********  -->
  <group ns="$(arg robot_2)">
    <param name="tf_prefix" value="$(arg robot_2)" />

    <!-- Map server -->
    <node name="map_server" pkg="map_server" type="map_server" args="$(arg map_file)">
       <param name="frame_id" value="map" /> 
    </node>

    <!-- **Turtlebot3** --> 
    <include file="$(find navigation_wx)/launch/robot_model.launch">
      <arg name="model" value="$(arg model)" />
      <!-- arg name="robot_name" value="$(arg robot_2)" --> 
    </include>
[... 55 lines cut ...]

  <group ns="$(arg robot_3)">
     <node pkg="navigation_wx" name="simple_navigation_goals" type="simple_navigation_goals" />
  </group>
</launch>
```

There are many warnings, like this in terminal:
[ WARN] [1546396297.896319112]: Timed out waiting for transform from tb3_2/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100459 timeout was 0.1.
[ WARN] [1546396297.941585960]: Timed out waiting for transform from tb3_3/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.10052 timeout was 0.1.
[ WARN] [1546396297.990269168]: Timed out waiting for transform from tb3_4/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100729 timeout was 0.1.
[ WARN] [1546396302.924653309]: Timed out waiting for transform from tb3_2/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100198 timeout was 0.1.
[ WARN] [1546396302.973580408]: Timed out waiting for transform from tb3_3/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100244 timeout was 0.1.
[ WARN] [1546396303.019011502]: Timed out waiting for transform from tb3_4/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100783 timeout was 0.1.
[ WARN] [1546396307.360798518]: No laser scan received (and thus no pose updates have been published) for 1546396307.360766 seconds.  Verify that data is being published on the /tb3_2/scan topic.
[ WARN] [1546396307.380160518]: No laser scan received (and thus no pose updates have been published) for 1546396307.380122 seconds.  Verify that data is being published on the /tb3_3/scan topic.
[ WARN] [1546396307.382364806]: No laser scan received (and thus no pose updates have been published) for 1546396307.382334 seconds.  Verify that data is being published on the /tb3_4/scan topic.
[ WARN] [1546396307.954685441]: Timed out waiting for transform from tb3_2/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101053 timeout was 0.1.
[ WARN] [1546396308.008738056]: Timed out waiting for transform from tb3_3/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101109 timeout was 0.1.
[... cut ...]
[ WARN] [1546396313.037830253]: Timed out waiting for transform from tb3_3/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100037 timeout was 0.1.
[ WARN] [1546396313.076757104]: Timed out waiting for transform from tb3_4/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100019 timeout was 0.1.

kijongGil, 2019-01-02:
This log is occurs when TF is incorrect. You can check TF tree in rqt. You have to set TF tree as map -> tb3_#/odom -> tb3_#/base_footprint -> .... 
Refer to [this](https://github.com/ROBOTIS-GIT/open_manipulator_with_tb3/blob/master/open_manipulator_with_tb3_tools/launch/open_manipulator_with_tb3_navigation.launch)

wx0205, 2019-01-02:
Thank you very much! I work hard on this, however, I have no idea how to realize the TF tree. Can you give me some advise in detail?  Really appreciate for your great help.

kijongGil, 2019-01-03:
You can run the TF tree with the following command.
`rosrun rqt_tf_tree rqt_tf_tree`

The original TF tree of turtlebot3 is map -> odom -> base_footprint -> etc....
But, If you want to run multi navigation, you have to make each name space as like tb3_1, tb3_2, etc... and then add to group ns in each turtlebot3_remote.launch.

wx0205, 2019-01-08:
I have got the tf : map -> tb3_#/odom -> tb3_#/base_footprint -> tb3_#/base_link, however, when I send goals in RVIZ, the turtlebot3 can't reach the goal and can't avoid  obstacle.  Can you give me some advise? Thank you !

wx0205, 2019-01-10:
I use the map  reconstructed by SLAM, and I use "map"  to be the frame_id to send goals by my codes or send goals in RVIZ,  turtlebot3 can move but can't reach goal and avoid obstacle. Maybe the frames can't transform correctly. When I check the nodes and topics, the move_base didn't subscribe the scan, but I can't modify it. So maybe I need your help. I am very grateful for your help!

kijongGil, 2019-01-17:
I'm sorry for late reply.
How did you use navigation param files? If you want to run three turtlebot3, you need to three navigation param files. and modified each namespace/scan.
For example, In costmap_common_params_burger.yaml, there is scan param. you have to modified sensor_frame from base_scan to tb3_#/base_scan and topic from scan to tb3_#/scan.
Beside amcl.launch also modifed to adding namespace scan, odom and base_footprint.

kijongGil, 2019-01-24:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-01-24 by kijongGil

ho0-kim, 2019-09-16:
> I have got the tf : map -> tb3_#/odom -> tb3_#/base_footprint -> tb3_#/base_link, however, when I send goals in RVIZ, the turtlebot3 can't reach the goal and can't avoid obstacle. Can you give me some advise? Thank you !

@wx0205 
Hi, I'm struggling for solving the same issue that there is no map topic in tf tree when I tried multiple turtlebot spawning.
Can you share your code or solution on it?
I want to get some help from you. It would be appreciated so much.

Jason-Zhou1, 2020-04-08:
Hi wx0205, I have seen your comment on "Jan 2, 2019".
I want to launch a file similar to your navigation_mutli.launch, but i don't know the content of "multi_amcl.launch" and "move_base.launch". Especially, i don't know if there are any more items besides odom and base_footprint that need to be put into namespace. Can you please show me your files and give me some advice?
Thank you!

wx0205, 2020-04-08:
I'm sorry. I can't show the files now because they don't exist in my pc. In my opinion, in the file of navigation_multi.launch, you need include the files of amcl and  move_base  of each robot in different namespaces. And you can use the tree of nodes to see all nodes in different namespaces.

Jason-Zhou1, 2020-04-08:
Thanks wx0205 for rely.
I have written a amcl launch file for the robot_1 as following:
<launch>
    <node pkg="amcl" type="amcl" name="amcl" output="screen">
        <param name="base_frame_id" value="robot_1/base_footprint"/> 
        <param name="gui_publish_rate" value="10.0"/> 
        <param name="kld_err" value="0.05"/>
        <param name="kld_z" value="0.99"/>
        <param name="laser_lambda_short" value="0.1"/>
        <param name="laser_likelihood_max_dist" value="2.0"/>
        <param name="laser_max_beams" value="60"/>	
        <param name="laser_model_type" value="likelihood_field"/>
        <param name="laser_sigma_hit" value="0.2"/>
        <param name="laser_z_hit" value="0.5"/>
        <param name="laser_z_short" value="0.05"/>
        <param name="laser_z_max" value="0.05"/>
        <param name="laser_z_rand" value="0.5"/>
        <param name="max_particles" value="2000"/>
        <param name="min_particles" value="500"/>
        <param name="odom_alpha1" value="0.25"/> 
        <param name="odom_alpha2" value="0.25"/> 
        <param name="odom_alpha3" value="0.25"/> 
        <param name="odom_alpha4" value="0.25"/> 
        <param name="odom_alpha5" value="0.1"/> 
        <param name="odom_frame_id" value="robot_0/odom"/>
        <param name="odom_model_type" value="diff"/>
        <param name="recovery_alpha_slow" value="0.001"/> 
        <param name="recovery_alpha_fast" value="0.1"/> 
        <param name="resample_interval" value="1"/> 
        <param name="transform_tolerance" value="1.25"/> 
        <param name="update_min_a" value="0.2"/> 
        <param name="update_min_d" value="0.2"/> 
    </node>
</launch>

Is there any problem for this file? Why can't i move the robot using "2D Nav Goal" button in Rviz?

Jason-Zhou1, 2020-04-08:
<launch>
    <node pkg="amcl" type="amcl" name="amcl" output="screen">
        <param name="base_frame_id" value="robot_0/base_footprint"/> 
        <param name="gui_publish_rate" value="10.0"/> 
        <param name="kld_err" value="0.05"/>
        <param name="kld_z" value="0.99"/>
        <param name="laser_lambda_short" value="0.1"/>
        <param name="laser_likelihood_max_dist" value="2.0"/>
        <param name="laser_max_beams" value="60"/>	
        <param name="laser_model_type" value="likelihood_field"/>
        <param name="laser_sigma_hit" value="0.2"/>
        <param name="laser_z_hit" value="0.5"/>
        <param name="laser_z_short" value="0.05"/>
        <param name="laser_z_max" value="0.05"/>
        <param name="laser_z_rand" value="0.5"/>
        <param name="max_particles" value="2000"/>
        <param name="min_particles" value="500"/>
        <param name="odom_alpha1" value="0.25"/> 
        <param name="odom_alpha2" value="0.25"/> 
        <param name="odom_alpha3" value="0.25"/> 
        <param name="odom_alpha4" value="0.25"/> 
        <param name="odom_alpha5" value="0.1"/> 
        <param name="odom_frame_id" value="robot_0/odom"/>
        <param name="odom_model_type" value="diff"/>
        <param name="recovery_alpha_slow" value="0.001"/> 
        <param name="recovery_alpha_fast" value="0.1"/> 
        <param name="resample_interval" value="1"/> 
        <param name="transform_tolerance" value="1.25"/> 
        <param name="update_min_a" value="0.2"/> 
        <param name="update_min_d" value="0.2"/> 
    </node>
</launch>

Jason-Zhou1, 2020-04-08:
The content of the launch file is attached. Thanks!
[amcl_1.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/4448496/amcl_1.txt)

Jason-Zhou1, 2020-04-08:
I also attached the move_base1.launch for robot_1. I don't know if i need to change /cmd_vel into robot_1/cmd_vel.
[move_base1.launch.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/4449065/move_base1.launch.txt)

How can i modify the amcl and move_base launch files so that i  drive the multi robots by clicking "2D Nav Goal" button in Rviz?
Thanks in advance!

ROBOTIS-Will, 2020-04-09:
@Jason-Zhou1 
You should also change the cmd_vel so that specific topic can be delivered to designated robot.
The `move_base` launch file loads costmap parameters, and it is recommended to create separate yaml files for each turtlebot such as `local_costmap_params_robot_1.yaml`.
In the costmap param files, all the frame and topic should refer to each robot.
In case of local_costmap_params.yaml,
```
local_costmap:
  global_frame: /robot_1/odom
  robot_base_frame: /robot_1/base_footprint
...
```
However, the global frame in the global costmap should be using `map` as its frame id as below.
```
global_costmap:
  global_frame: map
  robot_base_frame: /robot_1/base_footprint
...
```
The costmap_common_params_burger file should also be modified as
```
...
scan: {sensor_frame: /robot_1/base_scan, data_type: LaserScan, topic: /robot_1/scan, marking: true, clearing: true}
```
Please also refer to this issue #517 
Thank you.

Jason-Zhou1, 2020-04-09:
@ROBOTIS-Will 
Thank you very much for your advice!

Zuhair-A-Ahmed, 2024-04-15:
> @Jason-Zhou1 You should also change the cmd_vel so that specific topic can be delivered to designated robot. The `move_base` launch file loads costmap parameters, and it is recommended to create [...]

Dear @ROBOTIS-Will,
After several days of trying to solve issues of real multi turtlebot3 demo. Ive not seen any issue has been solved completely.  Still al the issue of :
1. Base to Map frames 
2. Laser scanner issue (conflict between real data sensors(

Would you please clearly inform us if there is no way to demonstrate real mutli turtlebot3 project at all ?

Thanks and regards,