ROBOTIS-GIT/turtlebot3 #364: wheel_left_link and wheel_right_link not working in rviz

extrasleepy, 2019-01-03:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [X ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ X] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [X ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.12.13] 
    - Firmware version: [1.2.3]
 
6. Write down the commands you used in order

$ export TURTLEBOT3_MODEL=burger
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz

7. Copy and Paste your error message on terminal

No problem in terminal
  
8. Please, describe detailedly what difficulty you are in 

 My turtlebot is not placing the wheels correctly in the rviz program. I see a Status Error under the robot model and the wheel_left_link and wheel_right_link are red
[image]
[image]

The turtlebot will move forward correctly when I press the test buttons on the openCR board but when I open rviz, the model has the wheels under the robot???

I see others have had this problem however I'm certain that I have the correct IP addresses.

Thank you for any advice!

kijongGil, 2019-01-04:
Hi @extrasleepy :)
This issue is probably a connection issue with OpenCR.
Refer to [this](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#programmer-setting) and press reset button.

Thanks,
Gilbert.

kijongGil, 2019-01-04:
In this picture, your master IP is 127.0.0.1, right?
127.0.0.1 means localhost, so you can't connect to the remote PC.
What did you use network? Could you show me the terminal when you command ifconfig and ROS_MASTER_URI and ROS_HOSTNAME in bashrc file?.

kijongGil, 2019-01-04:
Did you use virtual Machine for remote PC? VM can't connect to wireless lan.
Your turtlebot3 is connect to wireless IP 192.168.1.84. So, your remote PC have to connecting same network(192.168.1.xxx).
I recommand that do not using VM.

extrasleepy, 2019-01-04:
Yes Gilbert, that was the problem. I figured out how to share the ip from the main machine to the VM and not everything is working. Thank you for the help!!!

closed 2019-01-04 by extrasleepy

kijongGil, 2019-01-04:
Good! If you have any question, feel free to ask. Thanks.

ubtjoe, 2019-07-01:
@extrasleepy I think I am having the same issue while trying to run the remote pc inside of a docker container.  What was your fix, specifically?  Please fill in answers for the questions below:

On turtlebot:
*ROS_MASTER_URI*=?
*ROS_HOSTNAME*=?

On remote pc:
*ROS_MASTER_URI*=?
*ROS_HOSTNAME*=?

Thanks

extrasleepy, 2019-07-01:
@ubtjoe I just didn't have the correct IPs. With the correct IPs on both machines this configuration will work:
http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#network-configuration