ROBOTIS-GIT/turtlebot3 #365: My robot do not move with cmd_vel sent

angrysword, 2019-01-04:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - HERE

routiful, 2019-01-04:
I can't help you.
Please write down the issue sheet in detail.

kijongGil, 2019-01-04:
Hi @angrysword :)
I think that this issue is OpenCR firmware. Could you try to update OpenCR using Arduino?
Refer to [this](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#arduino-ide).

Thanks,
Gilbert.

angrysword, 2019-01-04:
Hi @kijongGil , 

     I have updated the firmware, bringout is good now as it was. However, the problem is still there. It did not move the robot after I pub some cmd_vel. The below is diagnostics info . I think it did not find the motor. 

header:
  seq: 42
  stamp:
    secs: 1546639673
    nsecs: 170895867
  frame_id: ''
status:
  -
    level: 0
    name: "IMU Sensor"
    message: "Good Condition"
    hardware_id: "MPU9250"
    values: []
  -
    level: 0
    name: ''
    message: ''
    hardware_id: ''
    values: []
  -
    level: 0
    name: ''
    message: ''
    hardware_id: ''
    values: []
  -
    level: 0
    name: ''
    message: ''
    hardware_id: ''
    values: []
  -
    level: 0
    name: ''
    message: ''
    hardware_id: ''
    values: []

angrysword, 2019-01-07:
Hi, 
     I have tested the problem one to let it drive another openCR board turtle3_robot. I found it works very well.    Now it is very clear, the problem is openCR board. However, what I did wrong for the opencr board. Please Help me out. thanks

angrysword, 2019-01-07:
@routiful @kijongGil any update?

routiful, 2019-01-09:
Sorry for late reply.
Would you try to scan all Dynamixels after upload DYNAMIXEL-WORKBENCH example?
I attached reference how to upload the firmware.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#o_find_dynamixel

angrysword, 2019-01-17:
did find any thing.
[image]

kijongGil, 2019-01-17:
I think your OpenCR is broken. Please contact to our customer service.
contactus2@robotis.com

Thanks.

angrysword, 2019-01-17:
thank you very much

kijongGil, 2019-01-18:
I'm going to close this issue. If you have any question, please ask me.
Thanks.

closed 2019-01-18 by kijongGil