ROBOTIS-GIT/turtlebot3 #366: Unable to see OpenCR board as /dev/ttyACM0

mkhansen-intel, 2019-01-05:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ x] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ x] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

I am unable to update the OpenCR firmware following the instructions here:
http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#ros2

The /dev/ttyACM0 device is no longer showing up. I have tried to go into FIrmware Recover Mode using these instructions, and I see the LED blinking but that doesn't make the device show up. 
http://emanual.robotis.com/docs/en/parts/controller/opencr10/#update-bootloaderlinux

I also tried booting into the bootloader mode, which makes it so I can see the device using lsusb, but still can't connect to it.  I'm not sure if I need to burn a new bootloader? 

Please advise.

OpusK, 2019-01-05:
Hi, @mkhansen-intel 

Please make sure that OpenCR is connected to another port. (eg. /dev/ttyACM1 ...)
The description of the manual assumes that the port of OpenCR is /dev/ttyACM0.

> I'm not sure if I need to burn a new bootloader?

Since we have not updated the boot loader, you do not need to update the boot loader separately.

mkhansen-intel, 2019-01-07:
When I look at the list of tty devices, there is no /dev/ttyAMC0 or /dev/ttyACM1.

mkhansen-intel, 2019-01-07:
I have tried unplugging / replugging to different USB ports and that made no difference.

OpusK, 2019-01-08:
@mkhansen-intel 

Please check if the device is recognized in [Firmware Recovery Mode](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#firmware-recovery-mode) 

And, are you currently testing on the Joule board?
If so, check it out on a desktop or laptop PC.

mkhansen-intel, 2019-01-09:
I was able to get the board to be recognized again by going into DFU mode then back into Firmware Recovery Mode. 

I'm now running the MicroXRCEAgent for both serial and udp. When I run the "turtlebot3_lidar" however I get a segmentation fault on the udp agent. Any idea why that might be happening?

OpusK, 2019-01-09:
@mkhansen-intel ,

Have you installed the MicroXRCE Agent on the commit version as described in the E-manual?

There are currently bugs to run multiple agents. I've raised [a issue with this](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/41), and it's still only specific branch commits.

So, the commit version of the Agent is very important :)

mkhansen-intel, 2019-01-09:
I did check out what the directions here said:
http://emanual.robotis.com/docs/en/popup/turtlebot3_ros2_sbc_setting
```
turtlebot@turtlebot:~/Micro-XRCE-DDS-Agent$ git status
HEAD detached at 61dc6ab
nothing to commit, working directory clean
turtlebot@turtlebot:~/Micro-XRCE-DDS-Agent$ 
```
And here's the client:
```
turtlebot@turtlebot:~/Micro-XRCE-DDS-Client$ git status
HEAD detached at v1.0.1
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   CMakeLists.txt
	modified:   client.config

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	examples/turtlebot3_lidar.tar.bz2
	examples/turtlebot3_lidar/

no changes added to commit (use "git add" and/or "git commit -a")
turtlebot@turtlebot:~/Micro-XRCE-DDS-Client$ 
```

Here is the error I am seeing:
```
turtlebot@turtlebot:~/turtlebot3$ ./turtlebot3_lidar 
Error at create entities: participant: 255 topic: 255 publisher: 255 darawriter: 255
```
The dmesg is this:
```
[ 1762.254319] traps: MicroXRCEAgent[4622] general protection ip:7f37450eabe2 sp:7f37420a96c0 error:0 in libmicroxrcedds_agent.so.1.0.0[7f374507c000+b1000]
```

mkhansen-intel, 2019-01-09:
I re-downloaded and re-built everything with debug symbols and ran in gdb. The segfault is being caused when main.cpp:111 executes in the turtlebot_lidar program. I printed out some of the variables being used there to see if this offers any clues:
```
(gdb) run
Starting program: /home/turtlebot/turtlebot3/turtlebot3_lidar 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1, main (args=1, argv=0x7fffffffdaa8)
    at /home/turtlebot/Micro-XRCE-DDS-Client/examples/turtlebot3_lidar/main.cpp:111
111	  if(!uxr_run_session_until_all_status(&session, 1000, requests, status, 4))
(gdb) next
113	    printf("Error at create entities: participant: %i topic: %i publisher: %i darawriter: %i\n", status[0], status[1], status[2], status[3]);
(gdb) print requests
$2 = {10, 11, 12, 13}
(gdb) print status
$3 = "\377\377\377\377"
(gdb) print session
$4 = {info = {id = 129 '\201', 
    key = "\314\314\335", <incomplete sequence \335>, 
    last_requested_status = 0 '\000', last_request_id = 14}, streams = {
    output_best_effort = {{buffer = 0x0, writer = 0, size = 0, 
        offset = 0 '\000', last_send = 0}}, 
    output_best_effort_size = 0 '\000', output_reliable = {{
        buffer = 0x7ffffffed9a0 "\253\001", size = 32768, history = 8, 
        offset = 4 '\004', last_written = 1, last_sent = 0, 
        last_acknown = 65535, next_heartbeat_timestamp = 1547061078164, 
        next_heartbeat_tries = 9 '\t', send_lost = false}}, 
    output_reliable_size = 1 '\001', input_best_effort = {{last_handled = 0}}, 
    input_best_effort_size = 0 '\000', input_reliable = {{
        buffer = 0x7fffffff59a0 "", size = 32768, history = 8, 
        last_handled = 65535, last_announced = 65535}}, 
    input_reliable_size = 1 '\001'}, comm = 0x7ffffffed940, 
  request_list = 0x7ffffffebb40, 
  status_list = 0x7ffffffed970 "\377\377\377\377", 
  request_status_list_size = 0, on_status = 0x0, on_status_args = 0x0, 
  on_topic = 0x0, on_topic_args = 0x0}
(gdb) 
```
Let me know if this helps or if there is anything more I can give you via gdb?

OpusK, 2019-01-10:
I think we need to check this a bit more. 
Since the products of ROS2 and eProsima are still under development, it seems to be fluctuating.

There are currently similar issues. Would you like to refer to it once?
 - [OpenCR Binary with FastRTPS 1.7.0](https://github.com/ROBOTIS-GIT/OpenCR-Binaries/issues/2)

mkhansen-intel, 2019-01-11:
Hmm, so if I understand correctly I should check out the master branch and re-pull the MicroXRCE-DDS-Agent, rebuild it and try that. I will give that a try

mkhansen-intel, 2019-01-11:
OK, so good news, when I rebuilt and run the master branch, I no longer have the seg fault. However, I go to the next few steps and the ros2 topic list doesn't contain everything in the directions:
```
turtlebot@turtlebot:~$ ros2 topic list
/imu
/joint_states
/odom
/parameter_events
/robot_description
/scan
/scan_half
/sensor_state
/tf
/tf_static
/time_sync
turtlebot@turtlebot:~$
```
It looks like the /motor_power topic is missing, along with /cmd_vel and several others. Should those be published by the OpenCR firmware? I saw it flash correctly and I've reset the board several times, even rebooted, but don't see those topics being published. When I run the tele-op, it sends the correct /cmd_vel, but the wheels don't spin and of course the robot doesn't move. 

Is there a way to confirm the OpenCR firmware is working correctly? Should I see those topics in the list?

OpusK, 2019-01-14:
Hi, @mkhansen-intel 

I [updated e-manual](https://github.com/ROBOTIS-GIT/emanual/pull/641) today.
Please refer to the new e-manual and try again.

Note that when installing the agent, remove the "fastrtps" and "fastcdr" installed in sbc.

mkhansen-intel, 2019-01-18:
I was able to get the robot wheels to turn with the latest updates. I can operate the turtlebot3 with teleop now. However I'm unable to get google cartographer to build on Ubuntu 16.04, but it builds on my 18.04 desktop machine, so I'm going to try and upgrade the turtlebot3 to 18.04.

These issues are unrelated to my original issue, so I'm going to close this now. 

Thank you @OpusK for your help!

closed 2019-01-18 by mkhansen-intel