ROBOTIS-GIT/turtlebot3 #368: [Chapt. 13 Autonomous Driving]  line detection node not responding, maybe problem with turtlebot3_autorace_intrinsic_camera_calibration.launch

yuseidesu, 2019-01-07:
1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
    - sorry, I couldn't figure out how to check the version, but I have updated them few minutes ago according to 
http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#fa
so they should be the latest version.
    - I am using Raspberry Pi Camera Module V2 camera module (Daylight - element14) for camera

6. Write down the commands you used in order

    - So I followed the instruction mentioned in the autonomous driving sector, "Tutorials: 4.4. Settings for Recognition" and "Tutorials: 5.1 Lane Detection" 
http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#autonomous-driving

after completing all of the instructions on chapter 6,7,8,and ones up to this point on chapter 13 including camera calibration. (skipped chapters on SLAM and Simulation).
Just for ease of reading I have made a list below with an out of terminal instractions I followed intending to make the line follower work.

```
Tutorials: 4.4. Settings for Recognition

4.4.1 Remote PC Open new terminal, then enter
 $ roscore

4.4.2 TurtleBot SBC Open new terminal, then enter
 $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_camera_pi.launch

4.4.3 TurtleBot SBC Open new terminal, then enter
 $ export AUTO_IN_CALIB=action
 $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch

4.4.4 Remote PC Open new terminal, then enter
 $ export AUTO_EX_CALIB=action
 $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_extrinsic_camera_calibration.launch

Tutorials: 5.1 Lane Detection

5.1.1 Remote PC Open new terminal, then enter
 $ export AUTO_DT_CALIB=calibration
 $ roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_lane.launch

5.1.2 Remote PC Open new terminal, then enter
 $ rqt
clicking plugins -> visualization -> Image view on the top of the screen will make extra monnitor for camera view. Make 3 extra monitor in the rqt plate by following it. Then, choose /detect/image_yellow_lane_marker/compressed, /detect/image_lane/compressed and /detect/image_white_lane_marker/compressed topics on each of extra monitors.

5.1.3 Remote PC Open new terminal, then enter
 $ rosrun rqt_reconfigure rqt_reconfigure
then, adjust the parameter value in /camera/image_projection and /camera/image_compensation_projection that carries out visual modifications on the image. The parameter image_projection will change the shape of the red rectangle of /camera/image_extrinsic_calib/compressed image. Intrinsic camera calibration will transform the image surrounded by the red rectangle, and will show the image that looks from over the lane. After that, overwrite value on to the lane.yaml file in turtlebot3_autorace_detect/param/lane/. This will make the camera set its parameters as you set here from next launching.

5.1.4 Remote PC After overwriting the calibration file, close rqt_rconfigure and turtlebot3_autorace_detect_lane, then enter
 $ export AUTO_DT_CALIB=action
 $ roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_lane.launch

5.1.5 Remote PC
 $ roslaunch turtlebot3_autorace_control turtlebot3_autorace_control_lane.launch

5.1.6 TurtleBot SBC
 $ roslaunch turtlebot3_bringup turtlebot3_robot.launch

After entering these commands, the robot will kick-off to run.

```

7. Copy and Paste your error message on terminal

There are several Errors I have recieved, but I am not sure which one is fundemental problem so I will make a list.

- 4.1.3
This might be a minor problem, but here some of the node could not be launched.
    ```
    TurtleBot SBC

    $ export AUTO_IN_CALIB=action
    $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
    ... logging to /home/pi/.ros/log/4fbf5770-1272-11e9-84aa-001c42da1064/roslaunch-raspberrypi-2601.log
    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.

    started roslaunch server http://10.0.0.18:41425/

    SUMMARY
    ========

    PARAMETERS
     * /camera/image_proc/queue_size: 20
     * /rosdistro: kinetic
     * /rosversion: 1.12.13

    NODES
      /camera/
        image_proc (image_proc/image_proc)
      /
        republish (image_transport/republish)

    ROS_MASTER_URI=http://10.0.0.28:11311

    process[republish-1]: started with pid [2610]
    ERROR: cannot launch node of type [image_proc/image_proc]: image_proc
    ROS path [0]=/opt/ros/kinetic/share/ros
    ROS path [1]=/home/pi/catkin_ws/src
    ROS path [2]=/opt/ros/kinetic/share
    ```

- 5.1.3

Here is where the main problem is. when I do
```
$ rqt
```
it works fine but when I try to Image view the /detect/image_yellow_lane_marker/compressed, /detect/image_lane/compressed and /detect/image_white_lane_marker/compressed, all three does not show any image but instead show a gradation from white to gray. I can see the normal image from camera on /camera/image without any problem.

- 5.1.6
Here the robot does not move.

8. Please, describe detailedly what difficulty you are in 

    - My ultimage goal is to make this algorithm work and make robot run following the lines, but I am encountering small errors here and there and I am not sure where to tackle. My guess for the main problem is the error happening on SBC when I try to 
    ```
        $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
    ```
    This problem has been happening throughout chapter 13, and I could only calbrate camera by running same command on remote PC when the manual told me to do so on SBC.

    Could someone help me figure out how this error on 4.1.3 is happening and hopefully get my line trace cord working?

    Regards.

kijongGil, 2019-01-10:
Hi @yuseidesu :)
Did you install additional dependent packages?
If you are not, please type following command

`sudo apt-get install ros-kinetic-image-transport ros-kinetic-cv-bridge ros-kinetic-vision-opencv python-opencv libopencv-dev ros-kinetic-image-proc`

And then, If you have another issue about autonomous, please go to this [issue](https://github.com/ROBOTIS-GIT/turtlebot3_autorace).

Thanks,
Gilbert

yuseidesu, 2019-01-11:
Hi KijongGil, thank you so much for your support!
I have tried that before and I tried it again right now, but somehow the download can not be done on raspberry pi. (It does work on my remote pc.) Here is the error message I get when I type the command on raspberry pi. 

```
pi@raspberrypi:~ $ sudo apt-get install ros-kinetic-image-transport ros-kinetic-cv-bridge ros-kinetic-vision-opencv python-opencv libopencv-dev ros-kinetic-image-proc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-image-transport
E: Unable to locate package ros-kinetic-cv-bridge
E: Unable to locate package ros-kinetic-vision-opencv
E: Unable to locate package ros-kinetic-image-proc
```

kijongGil, 2019-01-17:
I'm sorry for late reply.
This problem is that your mirror site is wrong. Please refer to [this](https://raspberrypi.stackexchange.com/questions/51928/how-to-change-mirror-fast-and-near-site-from-archive-respibian) and change your mirror site.

kijongGil, 2019-01-24:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-01-24 by kijongGil

ahsanulkalam, 2019-03-17:
Hi @kijongGil . I faced the same problem as above. Already changed mirror site to Singapore but no luck. Any other possible solutions?

kijongGil, 2019-03-18:
Hi @ahsanulkalam :)
Did you use the Raspbian on Raspberry B or B+?
If so, I recommend that installing Ubuntu mate. Refer to [this](https://github.com/ROBOTIS-GIT/turtlebot3/issues/397).

Thanks,
Gilbert.

ahsanulkalam, 2019-03-18:
Ok i'll try it first and let you if it works. Thanks.