ROBOTIS-GIT/turtlebot3 #369: Version for Melodic and Ubuntu 18.04 timeline.

BowenY, 2019-01-15:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?
    - Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?
    - Raspberry Pi 3

3. Which OS you installed in SBC?
    - [ ] Raspbian

4. Which OS you installed in Remote PC?
Two:
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)

//5. Write down software version and firmware version
//6. Write down the commands you used in order
//7. Copy and Paste your error message on terminal
  
8. Please, describe detailedly what difficulty you are in:
    I saw someone asked similar question in Oct, 2018. In this case, I just would love to know is there a good timeline or assumption release time of TB3 for Melodic and Ubuntu 18.04 support? Thanks.

routiful, 2019-01-16:
Hi, Thank you for your inquery.
We are ready to release Melodic support. It will be end up to this month.

BowenY, 2019-01-16:
@routiful Got it, Thanks for the reply:)

mkhuthir, 2019-01-19:
@routiful Great News, Thank you very much to all ROBOTIS team.

robotpilot, 2019-01-22:
FYI, https://github.com/ros/rosdistro/pull/20016

robotpilot, 2019-03-05:
Released for ROS Melodic
http://repositories.ros.org/status_page/ros_melodic_default.html?q=turtlebot3

closed 2019-03-05 by robotpilot