ROBOTIS-GIT/turtlebot3 #376: [melodic] Robot model loading error both in Rviz and Gazebo

virgileTN, 2019-01-18:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?
    - [x ] I'm Trying to simulate them.
    - [x ] Burger
    - [x ] Waffle
    - [x ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [x] None cf: 1
    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?
    - [x] None cf: 1
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.14.3]  (melodic)
    - Firmware version: [x.x.x]

6. Write down the commands you used in order
### To install for melodic from sources

I'm using this script :
```
mkdir -p ~/catkin_ws/src/
cd ~/catkin_ws/src/
git clone https://github.com/ROBOTIS-GIT/turtlebot3.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_applications_msgs.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_applications.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3_autorace.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3_deliver.git
cd ~/catkin_ws/

sudo apt install ros-melodic-joy ros-melodic-teleop-twist-joy \
                 ros-melodic-teleop-twist-keyboard ros-melodic-laser-proc \
                 ros-melodic-rgbd-launch ros-melodic-depthimage-to-laserscan \
                 ros-melodic-rosserial-arduino ros-melodic-rosserial-python \
                 ros-melodic-rosserial-server ros-melodic-rosserial-client \
                 ros-melodic-rosserial-msgs ros-melodic-amcl \
                 ros-melodic-map-server ros-melodic-move-base \
                 ros-melodic-urdf ros-melodic-xacro \
                 ros-melodic-compressed-image-transport \
                 ros-melodic-rqt-image-view ros-melodic-gmapping \
                 ros-melodic-navigation ros-melodic-interactive-markers
# Result :
# E: Unable to locate package ros-melodic-teleop-twist-joy
# E: Unable to locate package ros-melodic-depthimage-to-laserscan
# E: Unable to locate package ros-melodic-gmapping

rosdep install --from-paths src --ignore-src -r -y
# Result :
# ERROR: the following packages/stacks could not have their rosdep keys resolved
# to system dependencies:
# turtlebot3_slam: Cannot locate rosdep definition for [gmapping]
# Continuing to install resolvable dependencies...
[... 134 lines cut ...]

  - There may be a problem in one model is causing the following error of quaternion conversion. Quaternions must be normed, if not that can create conversions error. A not defined quaternion aka (0,0,0,0) can lead to the following error ?

```
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/gazebo_ros/spawn_model", line 31, in <module>
    from tf.transformations import quaternion_from_euler
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 31, in <module>
    from .listener import Transformer, TransformListener, TransformerROS
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/listener.py", line 29, in <module>
    import numpy
  File "/home/virgile/.local/lib/python2.7/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
  File "/home/virgile/.local/lib/python2.7/site-packages/numpy/core/__init__.py", line 59, in <module>
    from . import numeric
  File "/home/virgile/.local/lib/python2.7/site-packages/numpy/core/numeric.py", line 3093, in <module>
    from . import fromnumeric
  File "/home/virgile/.local/lib/python2.7/site-packages/numpy/core/fromnumeric.py", line 17, in <module>
    from . import _methods
  File "/home/virgile/.local/lib/python2.7/site-packages/numpy/core/_methods.py", line 158, in <module>
    _NDARRAY_ARRAY_FUNCTION = mu.ndarray.__array_function__
AttributeError: type object 'numpy.ndarray' has no attribute '__array_function__'
```

Thanks for your help and beautiful work :)

virgileTN, 2019-01-21:
I solved the numpy problem, by uninstalling numpy with pip `pip uninstall numpy` (it was both installed by apt and pip). 

Currently, the gazebo sim is working fine with karto slam. But I'm still unable to see the model in fake node.

kijongGil, 2019-02-15:
I'm so sorry for late reply.
We will release soon ROS Melodic version with turtlebot3.
You will be able to work with Melodic. 

Thanks,
Gilbert.

virgileTN, 2019-02-19:
Don't worry, I successfully installed and runned it on several computers with a simple install script: 
```
mkdir -p ~/catkin_ws/src/
slam_turtlebot3_dir=$PWD
cd ~/catkin_ws/src/
if [ "`grep CATKIN_WS_DIR ~/.bashrc`" == "" ]
then
  echo 'export CATKIN_WS_DIR=~/catkin_ws' >> ~/.bashrc
  source ~/.bashrc
fi
if [ "`grep TURTLEBOT3_MODEL ~/.bashrc`" == "" ]
then
  echo 'export TURTLEBOT3_MODEL=burger' >> ~/.bashrc
  source ~/.bashrc
fi

git clone https://github.com/ROBOTIS-GIT/turtlebot3.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_applications_msgs.git --branch melodic-devel
git clone https://github.com/ROBOTIS-GIT/turtlebot3_applications.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3_autorace.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3_deliver.git

cd ~/catkin_ws/
# Install required packages
sudo apt install ros-melodic-joy \
                 ros-melodic-teleop-twist-keyboard \
                 ros-melodic-laser-proc \
                 ros-melodic-rgbd-launch \
                 ros-melodic-rosserial-arduino \
                 ros-melodic-rosserial-python \
                 ros-melodic-rosserial-server \
                 ros-melodic-rosserial-client \
                 ros-melodic-rosserial-msgs \
                 ros-melodic-amcl \
                 ros-melodic-map-server \
                 ros-melodic-move-base \
                 ros-melodic-urdf \
                 ros-melodic-xacro \
                 ros-melodic-compressed-image-transport \
                 ros-melodic-rqt-image-view \
                 ros-melodic-navigation \
                 ros-melodic-interactive-markers \
                 ros-melodic-slam-karto \
                 ros-melodic-cartographer \
                 ros-melodic-cartographer-ros \
                 ros-melodic-cartographer-ros-msgs \
                 ros-melodic-cartographer-rviz

# Install potentially missing dependencies
rosdep install --from-paths src --ignore-src -r -y

# Build the whole catkin workspace
catkin_make
```
It wasn't working because of new TF conventions in melodic where _/_ were suppressed to avoid confusion with topic names. So to make it work in melodic I just changed frames names removing slashes in `global_costmap_params.yaml` and `local_costmap_params.yaml` config files.

For example the new global_costmap_params.yaml is the following
```
global_costmap:
  global_frame: map
  robot_base_frame: base_footprint

  update_frequency: 10.0
  publish_frequency: 10.0
  transform_tolerance: 0.5

  static_map: true
```

closed 2019-02-19 by virgileTN