ROBOTIS-GIT/turtlebot3 #381: Can I use inlet Euclid on Turtlebot3? Has anyone done this before?

stevensu1838, 2019-01-23:
Planning buying a intel Euclid kit to replace the raspberry Pi? Have any of you done this before? Can I use the pre-installed ROS kinetic on Euclid kit with Turtlebot3? Does it (USB3.0)works well with OpenCR(USB2.0)? I need your confirmation to make the decision to buy one. Cheers

routiful, 2019-01-24:
Sorry, we don't  have any plan to do it.
But I assume that if Intel has support Euclid in ROS, you can try that what your purposes.

BruceLun, 2019-01-29:
@stevensu1838 
I have a Euclid. I found the installed ROS packages are basically "static" and cannot be updated. If you configure the sources.list per ROS instructions then pull pull packages from the repository per usual, you will kill the camera driver (a custom driver is being used). It gets overwritten by the general driver from the ROS distro. I've tried MANY different ways to make this work with no luck. Each time had to re-install the OS image to the Euclid to recover. If you want to manually install TB3 packages from source, that may work but there could be dependency issues due to the existing packages being a particular version. Since there was no easy way to upgrade ROS packages, I basically gave up. Battery didn't last very long either.

ROBOTIS-Kayman, 2019-06-10:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-06-10 by ROBOTIS-Kayman