ROBOTIS-GIT/turtlebot3 #383: turtlebot3 navigation costmap2DROS transform timeout and Could not get robot pose

velvetbleurose, 2019-02-01:
ISSUE TEMPLATE ver. 0.2.0

Set-up
- TurtleBot3: Waffle Pi
- SBC: Raspberry Pi 3
- OS installed in SBC: Raspbian
- OS installed on remote PC: Ubuntu 16.04 LTS
- SW version: v1.12.13
- Firmware version: v1.2.3
- Commands:
    [remote PC] _roscore_
    [turtlebot] _roslaunch turtlebot3_bringup turtlebot3_robot.launch_
    [remote PC] _roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping_
    [remote PC] _rosrun map_server map_saver -f map_
    [remote PC] kill slam
    [turtlebot] kill bringup
    [turtlebot] _roslaunch turtlebot3_bringup turtlebot3_robot.launch_
    [turtlebot] _roslaunch turtlebot3_navigation turtlebot_turtlebot3_navigation.launch map=$HOME/map.yaml_
 
- Problem:
I am trying to follow the navigation portion of the e-manual and have been running into a couple problems. After running the last command, I get the error shown below. The second error, "**could not get robot pose, cancelling reconfiguration**", comes and goes. I typically use the RC100 to move the bot as using the teleop keyboard doesn't work when trying to move the robot but works in simulation. 

[image]

I've tried redoing SLAM and trying navigation on the new map to see if maybe my map just isn't good but still get the same error. 

I've looked at other issues related to mine and have tried the proposed solutions (reseting OpenCR, restarting roscore and bringup, checking if turtlebot and the remote PC are on the same time using _ntpdate ntp.ubuntu.com_). None of these worked. 

I ran _roswtf_ to see what was wrong and noticed that the following nodes were not connected.

[image]

I noticed in the related issues that there were mentions of topic frequencies (tf and odom) so I'm attaching tf_monitor output. When I ran _rostopic hz /tf_, the average frequencies ranged from 30Hz to 70Hz. It also says it's an unknown publisher?

[image]
 
I also get this issue sometimes when bringing up turtlebot3: [WARN] Failed to get param: timeout expired

I'm unsure of what is going wrong and would really appreciate any help in getting to the bottom of this! Thank you so much in advanced!

kijongGil, 2019-02-01:
Hi, @velvetbleurose :)
To move the turtlebot using RC100 and teleop is absolutely different. The RC100 only controls the DXL by directly connecting to openCR. Teleop is example package for ROS. So, If you can't move to turtlebot3 using teleop, It is not connect to turtlebot3 and PC. This mean that you can't use another package.
First, you have to move the turtlebot3 using teleop.

kijongGil, 2019-02-01:
You can check connection turtlebot3 and remote PC by rostopic.
1. Bring up your turtlebot3.
2. Run teleop launch.
3. Check topic information follow command.
`rostopic info /cmd_vel`

If you connected, you can see publishers and subscribers nodes.

Actually, you can run the another package as SLAM, navigation. But, It can't move to turtlebot3.
In navigation, It use ROS message as /cmd_vel for moving turtlebot3. So, you can't work navigation.
But, In SLAM, It doesn't matter how to communicate for moving turtlebot3. Because It not use /cmd_vel for moving turtlebot3.

velvetbleurose, 2019-02-01:
I found the problem with regards to teleoperation - it was my OS firewall settings. I had only allowed turtlebot to communicate on port 11311 instead of allowing all connections. 

For this I ran `sudo ufw allow from turtlebot_ip`

However, my navigation problem still remains. I ran roswtf and get the same error outputs (move_base --> move_base not connected but should be) minus the cmd_vel error. I also still get the Costmap2DROS transform timeout error and I notice that tf_monitor still says it's from an unknown publisher.

kijongGil, 2019-02-11:
I'm sorry for late reply.
Did you type follow command in turtlebot3?
`roslaunch turtlebot3_navigation turtlebot_turtlebot3_navigation.launch map=$HOME/map.yaml`

This package has to run in remote PC.

velvetbleurose, 2019-02-11:
Yes, I am running it from remote PC. Both problems still persist, shown below. I just ran these right now.

[image]
[image]

kijongGil, 2019-02-15:
I'm sorry for late reply.
Did you try to sync time? Refer to [this](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc).
Type the following command in turtlebot3 and RemotePC.

`sudo ntpdate ntp.ubuntu.com`

velvetbleurose, 2019-02-18:
@kijongGil 

I have done the time sync numerous times. 

I got navigation to finally work by increasing the transform tolerance parameter and TurtleBot is able to navigate well so I fixed that issue! 

I still get the error that the two nodes that should be connected are not. Is that a major problem or if is it one I can ignore?

kijongGil, 2019-02-19:
Don't care, It is not problem.
Increasing the transform tolerance parameter is not good way. Because the transform tolerance is that how many different ros time btw Remote PC and turtlebot3. For example, the transform tolerance is 0.5, you get warning log when ros time btw Remote PC and turtlebot3 is over 0.5s. 

So, I recommend return to the transform tolerance. and update ntpdate frequently.

Thanks.

gpwhd07, 2019-02-27:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-02-27 by gpwhd07

RNANDEZ22, 2021-03-29:
did you solve it? I have exactly the same issue

DeMischi, 2022-03-22:
For future reference:

Had the same problem, syncing the time on the Turtlebot as well as on the remote PC fixed the problem using this:

$ sudo apt-get install ntpdate
$ sudo ntpdate ntp.ubuntu.com