ROBOTIS-GIT/turtlebot3 #385: turtlebot3 navigation & cartographer launch

Song-MS, 2019-02-07:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ O ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ O ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ O ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ O ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - ros2 launch turtlebot3_cartographer cartographer.launch.py
    - ros2 launch turtlebot3_navigation2 navigation2.launch.py

 
7. Copy and Paste your error message on terminal

ros2 launch turtlebot3_cartographer cartographer.launch.py
[ERROR] [launch.LaunchService]: Caught exception in launch (see debug for traceback): cannot import name 'DeclareLaunchArgument'

ros2 launch turtlebot3_navigation2 navigation2.launch.py
[ERROR] [launch.LaunchService]: Caught exception in launch (see debug for traceback): "package 'nav2_bringup' not found, searching: ['/opt/ros/bouncy', '/home/song/turtlebot3_ws/install/turtlebot3', '/home/song/turtlebot3_ws/install/turtlebot3_teleop', '/home/song/turtlebot3_ws/install/turtlebot3_node', '/home/song/turtlebot3_ws/install/turtlebot3_navigation2', '/home/song/turtlebot3_ws2/install/turtlebot3_navigation2', '/home/song/turtlebot3_ws/install/turtlebot3_msgs', '/home/song/turtlebot3_ws/install/turtlebot3_bringup', '/home/song/turtlebot3_ws/install/turtlebot3_description', '/home/song/turtlebot3_ws/install/turtlebot3_cartographer', '/home/song/turtlebot3_ws/install/pcl_conversions', '/home/song/ros2_ws/install/rviz_assimp_vendor', '/home/song/ros2_ws/install/ros_environment', '/home/song/ros2_ws/install/launch', '/home/song/ros2_ws/install/osrf_pycommon', '/home/song/ros2_ws/install/libyaml_vendor', '/home/song/ros2_ws/install/libcurl_vendor', '/home/song/ros2_ws/install/ament_cmake_gmock', '/home/song/ros2_ws/install/ament_cmake_gtest', '/home/song/ros2_ws/install/console_bridge_vendor', '/home/song/ros2_ws/install/ament_xmllint', '/home/song/ros2_ws/install/ament_pyflakes', '/home/song/ros2_ws/install/ament_pep8', '/home/song/ros2_ws/install/ament_pep257', '/home/song/ros2_ws/install/ament_lint_auto', '/home/song/ros2_ws/install/ament_cmake_pytest', '/home/song/ros2_ws/install/ament_cmake_nose', '/home/song/ros2_ws/install/ament_cmake_auto', '/home/song/ros2_ws/install/ament_cmake', '/home/song/ros2_ws/install/ament_cmake_test', '/home/song/ros2_ws/install/ament_cmake_target_dependencies', '/home/song/ros2_ws/install/ament_cmake_python', '/home/song/ros2_ws/install/ament_cmake_export_dependencies', '/home/song/ros2_ws/install/ament_cmake_libraries', '/home/song/ros2_ws/install/ament_cmake_include_directories', '/home/song/ros2_ws/install/ament_cmake_export_link_flags', '/home/song/ros2_ws/install/ament_cmake_export_interfaces', '/home/song/ros2_ws/install/ament_cmake_export_libraries', '/home/song/ros2_ws/install/ament_cmake_export_include_directories', '/home/song/ros2_ws/install/ament_cmake_export_definitions', '/home/song/ros2_ws/install/ament_cmake_core', '/home/song/ros2_ws/install/ament_package', '/home/song/ros2_ws/install/ament_lint_cmake', '/home/song/ros2_ws/install/ament_index_python', '/home/song/ros2_ws/install/ament_flake8', '/home/song/ros2_ws/install/ament_cpplint', '/home/song/ros2_ws/install/ament_cppcheck', '/home/song/ros2_ws/install/ament_copyright']

8. Please, describe detailedly what difficulty you are in 

    - I followed the instruction of turtlebot3 ros2 and it can worked until bringup and teleop and rviz.
      When launching turtlebot3 navigation & turtlebot3 cartographer, it came up like above.
      Firstly, I could not find what is related with "DeclareLaunchArgument" which written in turtlebot3 navigation.
     Secondly, is the file of "nav2_bringup" only limited in ros2 crystal version (Currently installed ros2 bouncy for RemotePC) ?  

    Thank you

routiful, 2019-02-08:
It looks like navigation2 packages did not installed in your workspace.
Please following below reference and install it correctly first.

https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md

mkhansen-intel, 2019-02-26:
@Song-MS - I don't believe this will work with 'Bouncy' release. You probably have to upgrade to Ubuntu 18.04 and ROS2 'Crystal'

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

closed 2019-08-21 by routiful