ROBOTIS-GIT/turtlebot3 #388: Raspberry Pi Burger Rasbian image compile error : Could not find the required component 'interactive_markers'.

yapbenzet, 2019-02-17:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

7. Copy and Paste your error message on terminal

    - when i use catkin_make command, I get this error message : 

Could not find the required component 'interactive_markers'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "interactive_markers" with any of the following names:

    interactive_markersConfig.cmake
    interactive_markers-config.cmake

  
8. Please, describe detailedly what difficulty you are in 

    -  I cannot install ros-kinetic-interactive-marker.  This raspberry pi image is downloaded from http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-based-on-raspbian this address.. So this is not a ubuntu mate distribution. I guess provided Linux distro image based on Raspbian has a compiled version of ros. so we cannot install interactive-marker.

sudo apt-get install ros-kinetic-interactive-marker
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ros-kinetic-interactive-marker

kijongGil, 2019-02-19:
Hi @yapbenzet :)
Yes, raspbian can't install ros-kinetic-interactive-marker package. So, In Our [E-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-based-on-raspbian), you have to remove some packages in turtlebot3

`sudo rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example`

These packages don't need to place in turtlebot3. Because these should be run in Remote PC.

Thanks,
Gilbert

Lloydf, 2019-02-19:
While this is the correct way to fix the rasbian image, If you wanted to down load packages to experiment with then this distro a copy of a pre-loaded image is not for you. Use the Ubuntu-Mate distro as it is not restrictive. At least it is now clear why they wanted them four pkgs removed in the build setup instructions. ;-)

yapbenzet, 2019-02-19:
Thank you very much. This is the solution of my problem... 
But I figure out the cause of this problem. I pulled the turtlebot3 project from github. So whenever i update the project from the github by git pull command, i will face with same problem.

> Hi @yapbenzet :)
> Yes, raspbian can't install ros-kinetic-interactive-marker package. So, In Our [...]

yapbenzet, 2019-02-19:
> While this is the correct way to fix the rasbian image, If you wanted to down load packages to experiment with then this distro a copy of a pre-loaded image is not for you. Use the Ubuntu-Mate [...]

I have two turtlebot 3 burger so i already have ubuntu mate with one.. But rasbian image is much better than ubuntu mate..

closed 2019-02-21 by yapbenzet