ROBOTIS-GIT/turtlebot3 #390: Port waffle / waffle pi URDF files to ROS2

mkhansen-intel, 2019-02-26:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ x] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x ] Intel NUC

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [x ] Ubuntu 18.04

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - N/A 
7. Copy and Paste your error message on terminal

    - N/A
  
8. Please, describe detailedly what difficulty you are in 

    - I'm trying to run the ros2 navigation2 on turtlebot3 on my TB3 waffle. I've replaced the Joule board with a NUC, installed and built all the software now. 

I am following the directions here: http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2

When I go to run, the directions are for the burger:
```
$ ros2 launch turtlebot3_bringup burger.launch.py
```
This will launch using the burger urdf file. Since the waffle is physically different, odometry will be different and it won't navigate correctly. 

So I'm requesting that the waffle urdf files be ported to ros2 and possibly that the launch file be changed to launch with either one. 

Please let me know if you can help and support this.

If not, if you could tell me of a good work-around, I can proceed that way also.

routiful, 2019-02-27:
I'm going to support waffle and waffle_pi.
It will be commit this week.

mkhansen-intel, 2019-02-27:
Thanks @routiful

routiful, 2019-02-28:
@mkhansen-intel 

I've modified launch file to support Waffle and Waffle_Pi
https://github.com/ROBOTIS-GIT/turtlebot3/commit/9dea76ec339ad029273195608031cfcb35a3cb35

You should check updated emanual
http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2

If you have further issue, please comments below

mkhansen-intel, 2019-03-03:
Everything launches now, closing the ticket.

Thanks @routiful

closed 2019-03-03 by mkhansen-intel