ROBOTIS-GIT/turtlebot3 #392: Feature request simulation of camera in gazebo

florianspy, 2019-03-02:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
Hello i would like to simulate the raspberry camera of the turtlebot in gazebo unfortunately i dont now where to put the necessary code in the urdf files. 
Even a camera similar to the raspberry pi camera would be of great help.
I think i would have to manipulate the two file turtlebot3_burger.gazebo.xacro  and turtlebot3_burger.urdf.xacro. But appending the with the following code didnt work out for me:
In turtlebot3_burger.urdf.xacro:

 <link name='cam_link'>
  <pose frame=''>-0.118317 0.002195 0.004205 -0 -0 -3.12703</pose>
  <inertial>
    <mass>0.1</mass>
    <inertia>
      <ixx>1</ixx>
      <ixy>0</ixy>
      <ixz>0</ixz>
      <iyy>1</iyy>
      <iyz>0</iyz>
      <izz>1</izz>
    </inertia>
    <pose frame=''>0 0 0 0 -0 0</pose>
  </inertial>
  <self_collide>0</self_collide>
  <kinematic>0</kinematic>
  <sensor name='camera1' type='camera'>
    <camera name="head">
      <horizontal_fov>1.3962634</horizontal_fov>
      <image>
        <width>800</width>
        <height>800</height>
        <format>R8G8B8</format>
      </image>
      <clip>
        <near>0.02</near>
        <far>300</far>
      </clip>
    </camera>
    <always_on>1</always_on>
    <update_rate>30</update_rate>
    <visualize>1</visualize>
    <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
    <alwaysOn>true</alwaysOn>
    <updateRate>0.0</updateRate>
    <cameraName>iris/camera</cameraName>
    <imageTopicName>image_raw</imageTopicName>
    <cameraInfoTopicName>camera_info</cameraInfoTopicName>
    <frameName>camera_link</frameName>
    <hackBaseline>0.07</hackBaseline>
    <distortionK1>0.0</distortionK1>
    <distortionK2>0.0</distortionK2>
    <distortionK3>0.0</distortionK3>
    <distortionT1>0.0</distortionT1>
    <distortionT2>0.0</distortionT2>
  </plugin>
  </sensor>
and in turtlebot3_burger.gazebo.xacro:
<plugin name="camera_plugin" filename="libgazebo_ros_openni_kinect.so">
          <baseline>0.2</baseline>
          <alwaysOn>true</alwaysOn>
          
          <updateRate>0.0</updateRate>
          <cameraName>camera_ir</cameraName>
          <imageTopicName>/camera/depth/image_raw</imageTopicName>
          <cameraInfoTopicName>/camera/depth/camera_info</cameraInfoTopicName>
          <depthImageTopicName>/camera/depth/image_raw</depthImageTopicName>
          <depthImageInfoTopicName>/camera/depth/camera_info</depthImageInfoTopicName>
          <pointCloudTopicName>/camera/depth/points</pointCloudTopicName>
          <frameName>camera_link</frameName>
          <pointCloudCutoff>0.05</pointCloudCutoff>
          <distortionK1>0</distortionK1>
          <distortionK2>0</distortionK2>
          <distortionK3>0</distortionK3>
          <distortionT1>0</distortionT1>
          <distortionT2>0</distortionT2>
          <CxPrime>0</CxPrime>
          <Cx>0</Cx>
          <Cy>0</Cy>
          <focalLength>0</focalLength>
          <hackBaseline>0</hackBaseline>
        </plugin>

florianspy, 2019-03-05:
Following the guide https://github.com/richardw05/mybot_ws/blob/master/src/mybot_description/urdf/mybot.gazebo i was able to create a camera for my turtlebot simulation

RyanJaehyunShim, 2019-03-13:
Dear florianspy, 

Sorry for the late response. Please open another issue if you have any other additional questions.

Good luck with work,
Ryan

closed 2019-03-13 by gpwhd07