ROBOTIS-GIT/turtlebot3 #393: Duration in getOdom is based on Time(0), results in huge odom errors

mkhansen-intel, 2019-03-03:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - HERE

mkhansen-intel, 2019-03-03:
I was able to fix this with this change:
```
robot@NUC1804:~/turtlebot3_ws/src/turtlebot3/turtlebot3$ git diff turtlebot3_node/src/odometry.cpp
diff --git a/turtlebot3_node/src/odometry.cpp b/turtlebot3_node/src/odometry.cpp
index be340e6..0216cd7 100644
--- a/turtlebot3_node/src/odometry.cpp
+++ b/turtlebot3_node/src/odometry.cpp
@@ -25,7 +25,7 @@ constexpr char CHILD_FRAME_ID_OF_ODOMETRY[] = "base_footprint";
 
 nav_msgs::msg::Odometry Odometry::getOdom(const rclcpp::Time now, const double wheel_radius)
 {
-  static rclcpp::Time last_time = rclcpp::Time(0);
+  static rclcpp::Time last_time = now;
   rclcpp::Duration duration(now.nanoseconds() - last_time.nanoseconds());    
   
   calcOdometry(duration, wheel_radius);
@@ -144,4 +144,4 @@ bool Odometry::calcOdometry(const rclcpp::Duration duration,
```
I can submit in a PR if you want.

routiful, 2019-03-04:
Please make PR to this branch
Thank you for contribution :)

mkhansen-intel, 2019-03-08:
@routiful - I haven't had time to submit this patch. It's a one-line change can you please just make it?

routiful, 2019-03-11:
I did commit including your suggestion.
Thanks you for your contribution.

mkhansen-intel, 2019-03-11:
Thanks for the fix, I'll close this issue.

closed 2019-03-11 by mkhansen-intel