ROBOTIS-GIT/turtlebot3 #394: cmd_vel topic doesn't exist

yechun1, 2019-03-07:
**Issue: Follow the guide(http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2) to use turtlebot3 on ROS2, execute bringup steps, and check the ros2 topic list,  the result is that no cmd_vel topic exist. Not known something if I have missed. Does anyone met the same issue before, would you please share some experience about how to fix or debug? many thanks.**

ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    -  Waffle

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Intel NUC6i

3. Which OS you installed in SBC?

    - Ubuntu 18.04

4. Which OS you installed in Remote PC?

5. Write down software version and firmware version

    - Software version: [master]
    - Firmware version: [1.0.1]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

Below is execution steps:
 rapdev1@rapdev1:~/turtlebot3$ MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit

rapdev1@rapdev1:~/turtlebot3$ MicroXRCEAgent udp 2018
UDP agent initialization... OK
Enter 'q' for exit
RTPS Participant matched 1.f.59.e.c6.16.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.d.aa.7f.0.0.1.0.0.0|0.0.1.c1

rapdev1@rapdev1:/data/proj/sdk/tb3_ws$ ~/turtlebot3/turtlebot3_lidar
turtlebot3 lidar client setup complete!
1 - Sent topic: 0.870000, range_max: 361
2 - Sent topic: 0.871000, range_max: 361
3 - Sent topic: 0.866000, range_max: 361

rapdev1@rapdev1:/data/proj/sdk/tb3_ws$ ros2 launch turtlebot3_bringup robot.launch.py
urdf_file_name : turtlebot3_waffle.urdf
[INFO] [launch]: process[robot_state_publisher-1]: started with pid [27970]
[INFO] [launch]: process[turtlebot3_ros-2]: started with pid [27971]
Initialize urdf model from file: /data/proj/sdk/tb3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_waffle.urdf
Parsing robot urdf xml string.
Link base_link had 7 children
Link camera_link had 2 children
Link camera_depth_frame had 1 children
Link camera_depth_optical_frame had 0 children
Link camera_rgb_frame had 1 children
Link camera_rgb_optical_frame had 0 children
Link caster_back_left_link had 0 children
Link caster_back_right_link had 0 children
Link imu_link had 0 children
Link base_scan had 0 children
Link wheel_left_link had 0 children
Link wheel_right_link had 0 children
got segment base_footprint
got segment base_link
got segment base_scan
got segment camera_depth_frame
got segment camera_depth_optical_frame
got segment camera_link
got segment camera_rgb_frame
got segment camera_rgb_optical_frame
got segment caster_back_left_link
got segment caster_back_right_link
got segment imu_link
got segment wheel_left_link
got segment wheel_right_link
[INFO] [turtlebot3_node]: Init TurtleBot3 Node Main
Adding fixed segment from base_footprint to base_link
Adding fixed segment from base_link to camera_link
Adding fixed segment from camera_link to camera_depth_frame
Adding fixed segment from camera_depth_frame to camera_depth_optical_frame
Adding fixed segment from camera_link to camera_rgb_frame
Adding fixed segment from camera_rgb_frame to camera_rgb_optical_frame
Adding fixed segment from base_link to caster_back_left_link
Adding fixed segment from base_link to caster_back_right_link
Adding fixed segment from base_link to imu_link
Adding fixed segment from base_link to base_scan
Adding moving segment from base_link to wheel_left_link
Adding moving segment from base_link to wheel_right_link

rapdev1@rapdev1:/data/proj/sdk/tb3_ws$ ros2 topic list
/imu
/joint_states
/odom
/parameter_events
/robot_description
/rosout
/scan
/scan_half
/sensor_state
/tf
/tf_static
/time_sync

yechun1, 2019-03-07:
Just fixed the issue when change TB3 connection with SBC from USB3.0 port to USB2.0 port.

mkhansen-intel, 2019-03-07:
@yechun1 - I have seen this too, and the only way I have to fix it is to either reboot the OpenCR board (works sometimes) or push the reset button on the OpenCR board. 

See this section here:
http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#opencr-setup

I'm assuming you followed these directions already. In which case that's when I find that pushing reset again can fix this issue. Maybe @routiful has a better understanding of what is happening and why?

mkhansen-intel, 2019-03-07:
Also, have you made sure that /dev/ttyACM0 is showing up when you do an 'ls'?

If so, do you see the microcontroller when you do an 'lsusb'?

yechun1, 2019-03-08:
thanks @mkhansen-intel. Reset OpenCR board really help us. And we found the root cause that the `MicroXRCEAgent serial /dev/ttyACM0` does  publish topic when set ros domain id, for example
`export ROS_DOMAIN_ID=120 `

when we disable ROS_DOMAIN_ID in bashrc, the MicroXRCEAgent could work normally and publisher the cmd_vel topic.

This may have another issue when multi-users run tb3 in the same local network.

mkhansen-intel, 2019-03-08:
@yechun1 - you will need to make the change I mentioned in #393 in order to get correct odometry, so navigation can work.

yechun1, 2019-04-02:
Reported the issue on https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/66

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

This issue was solved to apply new firmware

closed 2019-08-21 by routiful