ROBOTIS-GIT/turtlebot3 #396: ERROR: cannot launch node of type [turtlebot3_example/turtlebot3_marker_server]: can't locate node [turtlebot3_marker_server] in package [turtlebot3_example]

AidanMech, 2019-03-11:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ X] Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
  
3. Which OS you installed in SBC?

    - [X ] Ubuntu MATE 16.04 or later

4. Which OS you installed in Remote PC?

    - [X ] Ubuntu 16.04 LTS (Xenial Xerus)

5. Write down software version and firmware version

    - Software version: [ROS 1.12.14]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    [Remote PC]
-roscore
-roslaunch turtlebot3_bringup turtlebot3_remote.launch
-roslaunch turtlebot3_example interactive_markers.launch
 
7. Copy and Paste your error message on terminal

    -ERROR: cannot launch node of type [turtlebot3_example/turtlebot3_marker_server]: can't locate node [turtlebot3_marker_server] in package [turtlebot3_example]

  
8. Please, describe detailedly what difficulty you are in 

  so Rviz and bringup works just fine, however the issue comes when i try to launch the interactive markers into Rviz. once i use that command, i receive the above error message. i dont know where the file is or how i access them. i just want to be able to navigate using interactive markers

ROBOTIS-Kayman, 2019-03-12:
Hi, 

you can find the package using command `roscd`.
please type the belowing.
```
roscd turtlebot3_example
```

And could you check the permissions of the node to see if it has execute permission?
if not, add teh execute permission please.
```
roscd turtlebot3_example/nodes
chmod +x turtlebot3_marker_server
```

AidanMech, 2019-03-12:
when i type roscd turtlebot3_example/nodes
it says no such file or directory

when i type roscd turtlebot3_example/nodes chmod +x turtlebot3_marker_server
it says
usage: roscd package
jump to target package

ROBOTIS-Kayman, 2019-03-12:
Did you follow the instruction for setup of remote pc?
This is [robotis e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup), could you follow the setting for remote pc?

If you download the source code correctly, type the command `rospack profile` and `roscd`.
```
rospack profile
roscd turtlebot3_example
```

AidanMech, 2019-03-12:
yes i followed the manual exactly. here is the output for the first line
Full tree crawl took 0.025471 seconds.
Directories marked with (*) contain no manifest.  You may
want to delete these directories.
To get just of list of directories without manifests,
re-run the profile with --zombie-only
-------------------------------------------------------------
0.021044   /opt/ros/kinetic/share
0.001674   /home/aidanmarch17/catkin_ws/src
0.000607   /home/aidanmarch17/catkin_ws/src/turtlebot3
0.000136 * /opt/ros/kinetic/share/OpenCV-3.3.1-dev
0.000037 * /opt/ros/kinetic/share/man
0.000030 * /opt/ros/kinetic/share/OpenCV-3.3.1-dev/haarcascades
0.000023 * /opt/ros/kinetic/share/doc
0.000012 * /opt/ros/kinetic/share/OpenCV-3.3.1-dev/lbpcascades
0.000008 * /opt/ros/kinetic/share/man/man1
0.000005 * /opt/ros/kinetic/share/doc/liborocos-kdl

ROBOTIS-Kayman, 2019-03-12:
Could you check the source file directly?
 - Go to the `turtlebot3_example`
```
cd ~/catkin_ws/src/turtlebot3/turtlebot3_example
```
 - check `nodes` folder
```
ls 
```
- If it exists, check the permission

AidanMech, 2019-03-12:
solved! the file was in the wrong directory file, so after moving it, all is sorted. moved from the secondary turtlebot3 folder back to within the source file

closed 2019-03-12 by AidanMech

isshe2628, 2024-02-16:
i have a same issue like this . When i give roslaunch FIFO_robots.launch it shows 
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_first_robot.py]: can't locate node [FIFO_first_robot.py] in package [turtlebot3_gazebo]
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_second_robot.py]: can't locate node [FIFO_second_robot.py] in package [turtlebot3_gazebo]
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_third_robot.py]: can't locate node [FIFO_third_robot.py] in package [turtlebot3_gazebo]
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_fourth_robot.py]: can't locate node [FIFO_fourth_robot.py] in package [turtlebot3_gazebo]
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_fifth_robot.py]: can't locate node [FIFO_fifth_robot.py] in package [turtlebot3_gazebo]
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_sixth_robot.py]: can't locate node [FIFO_sixth_robot.py] in package [turtlebot3_gazebo]
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_seventh_robot.py]: can't locate node [FIFO_seventh_robot.py] in package [turtlebot3_gazebo]
ERROR: cannot launch node of type [turtlebot3_gazebo/FIFO_eighth_robot.py]: can't locate node [FIFO_eighth_robot.py] in package [turtlebot3_gazebo]
ive tried copying these nodes in the turtlebot3_gazebo>src and alson gave permissions to all these files, but still shows me this error But when i give roslaunch turtlebot3_gazebo FIFO_robot.launch it shows 
[FIFO_robot.launch] is neither a launch file in package [turtlebot3_gazebo] nor is [turtlebot3_gazebo] a launch file name
The traceback for the exception was written to the log file

Please help me to solve this. Thanks in advance :)