ROBOTIS-GIT/turtlebot3 #397: cannot install ros package

liusanchuan, 2019-03-12:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [√ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [√ ] Raspberry Pi 3 PLUS
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [√ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [√ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.2.0]
    - Firmware version: []
 
6. Write down the commands you used in order

    - " apt-cache search ros-kinetic"  get nothing .
 
7. Copy and Paste your error message on terminal

    - nothing output
  
8. Please, describe detailedly what difficulty you are in 

    - I change the system of raspberry into Raspbian image that [you pre-made](http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-based-on-raspbian).But there is a big problem we cannot fix, the Raspbian cannot install the ros package .

kijongGil, 2019-03-12:
Hi @liusanchuan :)
The repository of the raspbian that we pre-made is not suitable your network. You have to change repository and update. 
Also, I recommend ubuntu mate 16.04

Thanks,
Gilbert

liusanchuan, 2019-03-12:
my sbc is raspberry 3b +.
but your emanul say:
http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#raspberry-pi-3-setup
NOTE: Raspberry Pi 3 B+ is available in TurtleBot3 Burger and Waffle Pi. If you use Raspberry Pi 3 B+, please refer to the following.

For Ubuntu MATE, it does not support the Raspberry Pi B+. So the first method described above can not be used. Instead, we recommend using a second method Install Linux based on Raspbian below.

kijongGil, 2019-03-12:
Yes, unfortunately ubuntu mate officially support the B but not B+.
The raspbian that we pre-made has old ROS packages. So, please refer to below to reinstall ROS.
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi
Or If you need to install few packages, you can install packages from source code on their github.

liusanchuan, 2019-03-12:
just now ,i re-install ROS with your link above, but the there still not have the `ros-kinetic `package.
i don't know how to do.
~~~shell
pi@raspberrypi:~ $ sudo apt-get install ros-
ros-actionlib-msgs           ros-desktop-full-dev         ros-perception               ros-simulators-lisp-dev
ros-base                     ros-desktop-full-lisp-dev    ros-perception-dev           ros-simulators-python-dev
ros-base-dev                 ros-desktop-full-python-dev  ros-perception-lisp-dev      ros-std-msgs
ros-base-lisp-dev            ros-desktop-lisp-dev         ros-perception-python-dev    ros-std-srvs
ros-base-python-dev          ros-desktop-python-dev       ros-robot                    ros-stereo-msgs
ros-cmake-modules            ros-diagnostic-msgs          ros-robot-dev                ros-tf2-msgs
ros-core                     ros-geometry-msgs            ros-robot-lisp-dev           ros-topic-tools-srvs
ros-core-dev                 ros-map-msgs                 ros-robot-python-dev         ros-trajectory-msgs
ros-core-lisp-dev            ros-message-generation       ros-roscpp-msgs              ros-visualization-msgs
ros-core-python-dev          ros-message-runtime          ros-rosgraph-msgs            ros-viz
ros-core-rosbuild-dev        ros-mk                       ros-sensor-msgs              ros-viz-dev
ros-desktop                  ros-move-base-msgs           ros-shape-msgs               ros-viz-python-dev
ros-desktop-dev              ros-nav-msgs                 ros-simulators
ros-desktop-full             ros-pcl-msgs                 ros-simulators-dev
pi@raspberrypi:~ $ sudo apt-get update
命中:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
命中:2 http://archive.raspberrypi.org/debian stretch InRelease
命中:3 http://packages.ros.org/ros/ubuntu stretch InRelease
正在读取软件包列表... 完成
pi@raspberrypi:~ $ sudo apt-cache search ros-kinetic
pi@raspberrypi:~ $
~~~

kijongGil, 2019-03-12:
Type flowing command and show me terminal.
`cat /etc/apt/sources.list.d/ros-latest.list`

liusanchuan, 2019-03-12:
it response：
`➜  ~ cat /etc/apt/sources.list.d/ros-latest.list
deb http://packages.ros.org/ros/ubuntu stretch main`

ROBOTIS-Kayman, 2019-03-13:
As you can see in [the wiki](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi), Raspberrypi is not able to install ros by command(`$ sudo apt-get install ros-kinetic-***`) like other things.

Please refer to the installation section of [the wiki](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi).
Following `rosinstall_generator` command is the name of the package to install.
If you want to install `uvc_camera`, you can create the rosinstall file to install this package as shown below.
```
$ rosinstall_generator uvc_camera --rosdistro kinetic --deps --wet-only --tar > kinetic-uvc_camera-wet.rosinstall
```
Then follow the rest.

liusanchuan, 2019-03-13:
OK ， I hope that you can give the accurate procedure in your e-manual web about how to use `Raspbian` on `Raspbarry 3B+`,
because  as  a agency ，we have sell out many Waffle pi with Rasp 3B+, while the Ubuntu meta cannot use and the Raspbian have no manual detialed.  OMG

ROBOTIS-Kayman, 2019-03-14:
There is an unofficial Ubuntu Mate 16.04 Image for Raspberrypi 3 B+.
When I tested it, it worked well. But we couldn't write down to the e-manual because it's unofficial.
This is a [Link](https://ubuntu-mate.community/t/ubuntu-mate-for-raspberry-pi-3-model-b/15867/8)

ROBOTIS-Kayman, 2019-04-12:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-04-12 by ROBOTIS-Kayman

liusanchuan, 2019-09-12:
Hi, I have the same problem. I use the Raspbian on raspberry 3b+ with open-maniplulator. And I can't install the Moveit on Raspberry Debian, Can you give me sone guidance.

rjshim, 2019-09-14:
@liusanchuan 

There is no need for you to use SBCs (i.e., raspberry) for OpenManipulator. We didn't even write anything about it on our [eManual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/overview). Just use a normal PC for it, and if you still have any questions about the usage, write them at the [link](https://github.com/ROBOTIS-GIT/open_manipulator/issues).

Regards,
Ryan