ROBOTIS-GIT/turtlebot3 #398: laser publisher still sending message if the sensor was disabled.

ahuizxc, 2019-03-15:
if the laser sensor was not launched, this topic publisher shouldn't send any topic, but now i got empty message. This problem caused error when i trying to slam by RGBD camera. thanks

RyanJaehyunShim, 2019-03-21:
@ahuizxc 

Sorry for the late response. To give us a clearer idea of the issue you have, you might want to fill in the below form that we ask for as a default. 

Regards,
Ryan 

--------------------------------------
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - HERE
---------------------------------------------------

ahuizxc, 2019-03-21:
My environment is ok, it's due to code logic. The laser publisher keep sending even if the sensor is not launched.

RyanJaehyunShim, 2019-03-21:
@ahuizxc 

We only support the basic tutorials introduced on our e-manual to help users with the initial setup and basic functions we provide. If the question is about your own program, we cannot give any help. If you encountered the issue while following our tutorial, please fill in the form and let us figure out the reason you are having that issue.

Kind regards,
Ryan

ROBOTIS-Kayman, 2019-04-12:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-04-12 by ROBOTIS-Kayman