ROBOTIS-GIT/turtlebot3 #399: How to change robot model while adding new component or replacing a existing one for turtlebot3 waffe?

gaoethan, 2019-03-19:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [X] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [X] etc (Intel NUC)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [X] etc (Ubuntu 18.04)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)
  
8. Please, describe detailedly what difficulty you are in 

The 360 LDS within turtlebot3 works well, but we plan to practice the depth_to_laserscan with depth image of RGB-D camera in order to replace the Laser Scan from the inherent LDS, moreover, we doesn't plan to use the R200 yet. However, once we try to install the new RGB-D camera on tutlebot3 waffe, the robot model (a.k.a URDF) should be updated for waffe, maybe there are 2 ways to go ahead:
- Option 1: put the new RGB-D camera to be installed at the position of the current LDS, but we want to keep the LDS, so this option is not the preference.
- Option 2: remove the R200 camera and put the new RGB-D camera at its position

For option 2, can we keep the original model description in the [turtlebot3_waffle.urdf.xacro](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_description/urdf/turtlebot3_waffle.urdf.xacro) for R200 and use it for the new RGB-D camera for its TF transformation?
As per my basic understanding, it should be incorrect with the same one, because the new RGB-D camera (e.g Realsene D435i) has different shape with the R200, their centroid shouldn't be at the same point. Could you please give some suggestion to enable right robot model description for the new camera on the current turtlebot3 waffe to allow the map with cartographer to be usable and accurate based on the correct transformation of laser scan from the camera depth ?
Basically, we think that we need to measure the shape of the new camera with some specific scale and make clear the internal link transformation of the camera, for example depth_link to camera_link... but to be frankly, not experienced with this, so please give some guidelines. thanks !

ROBOTIS-Kayman, 2019-03-21:
Hi.

 You don't need to measure the shape of the camera. the RGB-D ros package may provide its own TF tree. you have to make the robot model (URDF) including the base link of camera tf tree. Change the [urdf file](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_description/urdf/turtlebot3_waffle.urdf.xacro) from line 193 to line 249.  The important thing is a relative position between the links

 If you save the robot file to other name, you can keep the original. 
As you can see the [description file](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_bringup/launch/includes/description.launch.xml), we seperate the model using env parameter.
If you save to `turtlebot3_waffle_mod.urdf.xacro`, you can use it with the following.
```
$ export TURTLEBOT3_MODEL=waffle_mod
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz
```

please refer to this [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#load-a-turtlebot3-on-rviz).

gaoethan, 2019-03-21:
Thanks for your information @ROBOTIS-Kayman. 

Just as you said, if the RGB-D ros packages has its TF tree for its camera_link with the other internal unit, e.g depth_link and rgb_link, we can directly copy it to replace the line 193 to line 249 in the URDF, this is the easy case. But if there is no existing TF tree description for the RGB-D ros packages, how to deal with it and work out the corresponding internal camera TF tree ? from its spec or calibration by oneself?

Besides the above internal link TF tree with the camera_link, how to define the TF description between turtulebot3 waffe(base_link) and the camera_link of the new RGB-D camera is also necessary. For example, can I install the new RGB-D camera on the turtlebot3 waffe at any available place or replace its current R200 camera with its place ?  How to calculate the value btw the base_link and the camera_link for the case of any available place and the existing R200 place respectively?

Thanks

mkhansen-intel, 2019-03-22:
@gaoethan - I was able to get my Realsense D435 camera to work by commenting out the transforms below the camera_link level in the URDF file (lines 223-249), and use the realsense camera node which publishes the transforms from camera_link to the optical frames. That worked for me.

gaoethan, 2019-03-22:
@mkhansen-intel thanks for your comment. what I have is D435i, it's generally similar with D435 with one more inherent IMU. Besides workaround it for the current, I plan to learn some BKM to add and tune a new component for the robot model in different cases, especially for V-SLAM, it requires more precision.

ROBOTIS-Kayman, 2019-03-26:
To create urdf more accurately, you must use 3D CAD tools. After you have assembled the new model in the tool, you can measure the relative position between the base and the new component.

closed 2019-04-02 by gaoethan