ROBOTIS-GIT/turtlebot3 #400: Desktop PC as TurtleBot-SBC - Can't controll Dynamixel

philison, 2019-03-28:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Desktop PC (USB connection directly to OpenCR)

3. Which OS you installed in SBC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus) 

4. Which OS you installed in Remote PC?

    - [x] SBC = Remote PC
  
8. Please, describe detailedly what difficulty you are in 

    Hi,

i am working on a project where I want to replace the _Raspberry Pi 3_ of my _Turtlebot3 Waffle Pi_ with an Intel Nuc. But the Nuc wasn't delivered yet, so I am trying to use my Desktop PC as a replacement for the Pi.
The Desktop PC is running Windows 10 with Ubuntu 16.04 in an Oracle VM Virtual Box. 
I connectet the OpenCR via the USB cable to an USB-Port of the PC and gave the Virtual Box access to this USB-Port.
That is already enough to power up the OpenCR.
Than I went ahead and followed the instructions in the e-Manual. I installed everytghing that was ment for the Remote-PC and the TurtleBot (Pi), on my Desktop PC. I did not went through the SSH Setup, as my PC already is the TurtleBot-SBC.

So I went ahead and connected one Dynamixel to the OpenCR to see if it starts spinning when I run the following Teleoperaation - Keyboard commands.

[1]Terminal
$ roscore

[2]Terminal
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch

[3]Terminal
$ export TURTLEBOT3_MODEL=waffle_pi
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

Everything was running without errors and I was able to see changing Orientation-Values in the Topic-Monitor while turning the OpenCR on the table.
But the Dynamixel didn't start spinning while pressing 'W' in the [3]Terminal. (the values changed up to _linear vel=0.26_)

Rosgraph:
[image]
(I deactivated the LiDAR-Node in the launch file)

So I though it is because I haven't connected anny power source beside the USB-Connection. so I tried connecting the SMPS 12V 5A to the OpenCR and afterwards the Battery.  But still the Dynamixel wasn't moving.

Does someone has an idea whats wrong ?

Many thanks in advance!
Phil

RyanJaehyunShim, 2019-03-29:
@philison 

We do not recommend you use VM for the task. Our team has tested it already before and had the same issue. Please test it again with your new NUC PC and let us know if you encounter the same issue again.

Have a good day,
Ryan

philison, 2019-03-29:
Thank you very much for the quick reply @RyanJaehyunShim !
I'll try it on another PC with a normal Ubuntu install. 

Regards,
Phil

RyanJaehyunShim, 2019-04-01:
No problem. Please, open another issue if you have any additional questions. I will close this issue.

Regards,
Ryan

closed 2019-04-01 by RyanJaehyunShim