ROBOTIS-GIT/turtlebot3 #401: Node_option error during ROS2 Turtlebot3 Tutorial installation

Song-MS, 2019-04-04:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ o] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ o] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - mkdir -p ~/turtlebot3_ws/src

$ colcon build --symlink-install
**When building the turtlebot3_ws, the error came up as below** 

Starting >>> cv_bridge                                                 
--- stderr: image_transport                                                    
/home/user/turtlebot3_ws/src/gazebo/camera_info_manager/image_transport/test/test_remapping.cpp: In member function ‘virtual void TestPublisher::SetUp()’:
/home/user/turtlebot3_ws/src/gazebo/camera_info_manager/image_transport/test/test_remapping.cpp:22:13: error: ‘NodeOptions’ is not a member of ‘rclcpp’
     rclcpp::NodeOptions node_options;
             ^~~~~~~~~~~
/home/user/turtlebot3_ws/src/gazebo/camera_info_manager/image_transport/test/test_remapping.cpp:22:13: note: suggested alternative: ‘InitOptions’
     rclcpp::NodeOptions node_options;
             ^~~~~~~~~~~
             InitOptions
/home/user/turtlebot3_ws/src/gazebo/camera_info_manager/image_transport/test/test_remapping.cpp:23:5: error: ‘node_options’ was not declared in this scope
     node_options.arguments(arguments);
     ^~~~~~~~~~~~
/home/user/turtlebot3_ws/src/gazebo/camera_info_manager/image_transport/test/test_remapping.cpp:23:5: note: suggested alternative: ‘rcl_node_options_t’
     node_options.arguments(arguments);
     ^~~~~~~~~~~~
     rcl_node_options_t
make[2]: *** [CMakeFiles/image_transport-remapping.dir/test/test_remapping.cpp.o] Error 1

**I found this solution that ros2_ws install repos as master version(not crystal) which have node_options  because latest-version doesn't have node_options.**

https://github.com/ROBOTIS-GIT/emanual/issues/729

**but here is the another problem, which is the master version does not work to build.
(I guess the rosidl_generater_c file is some error)
I will try to figure out this error but I would like to know to build image_transport except  " node_options " in turtlebot3 installation.

(I wonder how to fix this problem in turtlebot3 installation, overcoming all these)  

Thank you**

EndlessLoops, 2019-04-08:
i also get the error

lalala@ubuntu:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> nav2_common
Starting >>> behaviortree_cpp
Finished <<< nav2_common [0.39s]                                   
Starting >>> nav2_msgs
Finished <<< behaviortree_cpp [1.60s]                                        
Starting >>> nav2_voxel_grid
Finished <<< nav2_msgs [1.81s]                                        
Starting >>> nav2_util                                      
Finished <<< nav2_voxel_grid [0.64s]
Starting >>> nav_2d_msgs
Finished <<< nav_2d_msgs [3.40s]                                   
Starting >>> nav2_robot
--- stderr: nav2_util                                              
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp: In function ‘rclcpp::Node::SharedPtr nav2_util::generate_internal_node(const string&)’:
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: error: ‘NodeOptions’ is not a member of ‘rclcpp’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: note: suggested alternative: ‘InitOptions’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
           InitOptions
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:69:3: error: ‘options’ was not declared in this scope
   options.use_global_arguments(false);
   ^~~~~~~
make[2]: *** [CMakeFiles/nav2_util_core.dir/src/node_utils.cpp.o] Error 1
make[1]: *** [CMakeFiles/nav2_util_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< nav2_util	[ Exited with code 2 ]
Aborted  <<< nav2_robot                                    

Summary: 5 packages finished [8.94s]
  1 package failed: nav2_util
  1 package aborted: nav2_robot
  1 package had stderr output: nav2_util
  49 packages not processed

EndlessLoops, 2019-04-08:
i also get the error

```
lalala@ubuntu:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> nav2_common
Starting >>> behaviortree_cpp
Finished <<< nav2_common [0.39s]                                   
Starting >>> nav2_msgs
Finished <<< behaviortree_cpp [1.60s]                                        
Starting >>> nav2_voxel_grid
Finished <<< nav2_msgs [1.81s]                                        
Starting >>> nav2_util                                      
Finished <<< nav2_voxel_grid [0.64s]
Starting >>> nav_2d_msgs
Finished <<< nav_2d_msgs [3.40s]                                   
Starting >>> nav2_robot
--- stderr: nav2_util                                              
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp: In function ‘rclcpp::Node::SharedPtr nav2_util::generate_internal_node(const string&)’:
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: error: ‘NodeOptions’ is not a member of ‘rclcpp’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: note: suggested alternative: ‘InitOptions’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
           InitOptions
/home/lalala/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:69:3: error: ‘options’ was not declared in this scope
   options.use_global_arguments(false);
   ^~~~~~~
make[2]: *** [CMakeFiles/nav2_util_core.dir/src/node_utils.cpp.o] Error 1
make[1]: *** [CMakeFiles/nav2_util_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< nav2_util	[ Exited with code 2 ]
Aborted  <<< nav2_robot                                    

Summary: 5 packages finished [8.94s]
  1 package failed: nav2_util
  1 package aborted: nav2_robot
  1 package had stderr output: nav2_util
  49 packages not processed
```

routiful, 2019-04-09:
@Song-MS @EndlessLoops 
Please downloads master branch of ROS2
```
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
```

The rclcpp have been updated.

EndlessLoops, 2019-04-09:
> @ Song-MS @EndlessLoops
> 请下载ROS2的主分支
> 
> ```
> wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
> ```
> rclcpp已更新。

I will continue to try today.
Thanks  for your reply.

EndlessLoops, 2019-04-09:
@routiful 
I will continue to try today.
Thanks for your reply.

EndlessLoops, 2019-04-09:
new building error
```
lalala1@ubuntu:~/turtlebot3_ws$ . ~/ros2_ws/install/local_setup.bash
lalala1@ubuntu:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> nav2_common
Starting >>> behaviortree_cpp
Finished <<< nav2_common [0.36s]                                   
Starting >>> nav2_msgs
Finished <<< behaviortree_cpp [1.87s]                                        
Starting >>> nav2_voxel_grid
Finished <<< nav2_voxel_grid [0.67s]                                  
Starting >>> nav_2d_msgs
Finished <<< nav2_msgs [2.56s]                                        
Starting >>> nav2_util
Finished <<< nav2_util [1.62s]                                         
Starting >>> nav2_dynamic_params
Finished <<< nav_2d_msgs [2.44s]                                        
Starting >>> nav2_robot                                      
Finished <<< nav2_dynamic_params [1.15s]                                       
Starting >>> dwb_msgs
Finished <<< nav2_robot [0.75s]                                        
Starting >>> nav2_tasks
--- stderr: nav2_tasks                                                        
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "Findbehaviortree_cpp.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "behaviortree_cpp", but CMake did not find one.

  Could not find a package configuration file provided by "behaviortree_cpp"
  with any of the following names:

    behaviortree_cppConfig.cmake
    behaviortree_cpp-config.cmake

  Add the installation prefix of "behaviortree_cpp" to CMAKE_PREFIX_PATH or
  set "behaviortree_cpp_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp" provides a separate development package or
  SDK, be sure it has been installed.

make: *** [cmake_check_build_system] Error 1
---
Failed   <<< nav2_tasks	[ Exited with code 2 ]
Aborted  <<< dwb_msgs                                           

Summary: 8 packages finished [8.77s]
  1 package failed: nav2_tasks
  1 package aborted: dwb_msgs
  1 package had stderr output: nav2_tasks
  46 packages not processed

```

routiful, 2019-04-09:
@EndlessLoops 
After behavior tree first,execute local_bash and build all packages.

```
$ colcon build --packages-select bahaviortree_cpp
$ . ~/turtlebot3_ws/install/local_setup.bash
$ colcon build --symlink-install
```

EndlessLoops, 2019-04-09:
@routiful
I will continue to try now.
Thanks for your rapid reply.

EndlessLoops, 2019-04-10:
@routiful 
[https://github.com/ROBOTIS-GIT/emanual/blob/master/docs/en/platform/turtlebot3/ros2.md](url)

Open CR setup

`$ wget https://github.com/ROBOTIS-GIT/OpenCR_Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.xz`

This link is incorrect, please edit

Song-MS, 2019-04-10:
@routiful 

I am sorry to ask but I cannot be clear this problem 

> wget  https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos

firstly, it should be built on ros2_ws folder. right?
but if I tried to this link, this error keep coming up as below

stderr: rosidl_generator_c                                        
Traceback (most recent call last):
  File "/home/song/ros2_ws/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 40, in <module>
    sys.exit(main())
  File "/home/song/ros2_ws/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 35, in main
    args.generator_arguments_file,
  File "/opt/ros/crystal/lib/python3.6/site-packages/rosidl_generator_c/__init__.py", line 45, in generate_c
    assert os.path.exists(template_file), 'Could not find template: ' + template_file
AssertionError: Could not find template: /home/song/ros2_ws/src/ros2/rosidl/rosidl_generator_c/resource/msg.h.em
make[2]: *** [rosidl_generator_c/rosidl_generator_c/msg/bool.h] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
make: *** [all] Error 2
---

How did you figure out this error, which is built from"wget  https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos"?

routiful, 2019-04-10:
@EndlessLoops 

Thank you! (https://github.com/ROBOTIS-GIT/emanual/pull/763)

routiful, 2019-04-10:
@Song-MS 
I can't pinpoint what exactly wrong. In my experience, To delete `ros2_ws` folders was help to successfully build it.

Song-MS, 2019-04-10:
@routiful 
I share to record the video which is the error when the state that is  "ros2_ws" deleted starts. 
https://drive.google.com/open?id=1yX5dT0YSVi1xLgF3U9oeIv7pxwrvwoCt 

Please check this link

EndlessLoops, 2019-04-11:
@routiful 
I also had [this problem](https://github.com/ROBOTIS-GIT/turtlebot3/issues/394).
I want to finish the demo of tb3-ros2 and how how should i fix the problem of teleop_keyboard?

```
ubuntu@tb3-ros2:~$ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit

ubuntu@tb3-ros2:~$ cd ~/turtlebot3 && MicroXRCEAgent udp 2018
UDP agent initialization... OK
Enter 'q' for exit

./turtlebot3/turtlebot3_lidar
turtlebot3 lidar client setup complete!
1 - Sent topic: 0.332000, range_max: 361
2 - Sent topic: 0.330000, range_max: 361
3 - Sent topic: 0.330000, range_max: 361
4 - Sent topic: 0.330000, range_max: 361
5 - Sent topic: 0.331000, range_max: 361
6 - Sent topic: 0.330000, range_max: 361
7 - Sent topic: 0.330000, range_max: 361
8 - Sent topic: 0.330000, range_max: 361
9 - Sent topic: 0.330000, range_max: 361
10 - Sent topic: 0.330000, range_max: 361
11 - Sent topic: 0.330000, range_max: 361
12 - Sent topic: 0.330000, range_max: 361
13 - Sent topic: 0.330000, range_max: 361

ubuntu@ubuntu-laptop:~$ ros2 launch turtlebot3_bringup robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2019-04-11-12-10-05-008511-ubuntu-laptop-10514
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [robot_state_publisher-1]: process started with pid [10523]
[INFO] [turtlebot3_ros-2]: process started with pid [10524]
[robot_state_publisher-1] Initialize urdf model from file: /home/ubuntu/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] got segment base_footprint
[robot_state_publisher-1] got segment base_link
[robot_state_publisher-1] got segment base_scan
[robot_state_publisher-1] got segment caster_back_link
[robot_state_publisher-1] got segment imu_link
[robot_state_publisher-1] got segment wheel_left_link
[robot_state_publisher-1] got segment wheel_right_link
[robot_state_publisher-1] Adding fixed segment from base_footprint to base_link
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_link
[robot_state_publisher-1] Adding fixed segment from base_link to imu_link
[robot_state_publisher-1] Adding fixed segment from base_link to base_scan
[robot_state_publisher-1] Adding moving segment from base_link to wheel_left_link
[robot_state_publisher-1] Adding moving segment from base_link to wheel_right_link
[turtlebot3_ros-2] [INFO] [turtlebot3_node]: Init TurtleBot3 Node Main

ubuntu@ubuntu-laptop:~/turtlebot3_ws$ ros2 topic list
/imu
/joint_states
/odom
/parameter_events
/robot_description
/rosout
/scan
/scan_half
/sensor_state
/tf
/tf_static
/time_sync
```

routiful, 2019-04-11:
@Song-MS 
I have seen attached video. The error msg represents that `rosidl_generator_c` was founded in ros2_install folders. I assume that the ros2_install folder might be deleted.
Moreover, I recommended that post your error in ROS answer to get another solution.

routiful, 2019-04-11:
@EndlessLoops 
That issue haven't solved yet but I aware of. It will be solved soon. Sorry for inconvenience.

miguelvelezmj25, 2019-04-11:
I am having the same "‘NodeOptions’ is not a member of ‘rclcpp’" issue when running ```colcon build --symlink-install``` inside ```~/turtlebot3_ws/```. Has that part been solved yet? (i.e., 15. 1.1.3 [Install TurtleBot3 ROS2 Packages] in the ROS2 Turtlebot3 tutorial)

heilund, 2019-04-14:
I am having the exact same problem as @miguelvelezmj25. Any solutions?

routiful, 2019-04-15:
@miguelvelezmj25  @heilund 

I already answer that question. Please check again my comments.

heilund, 2019-04-17:
After building behaviortree_cpp successfully with "colcon build --packages-select bahaviortree_cpp" i tried to execute ". ~/turtlebot3_ws/install/local_setup.bash", but i get the error: 
not found: "/home/robot/turtlebot3_ws/install/behaviortree_cpp/share/behaviortree_cpp/local_setup.sh"
not found: "/home/robot/turtlebot3_ws/install/behaviortree_cpp/share/behaviortree_cpp/local_setup.bash". 
Has anyone had this problem before?

miguelvelezmj25, 2019-04-18:
@routiful 

Which comment should fix the issue? For example, when you said to download the master branch of ROS2
```
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
```
when is that done? You also mentioned 

> After behavior tree first,execute local_bash and build all packages.
> 
> $ colcon build --packages-select bahaviortree_cpp
> $ . ~/turtlebot3_ws/install/local_setup.bash
> $ colcon build --symlink-install

I tried that, but I also got an error.

Which commands should I use to install the packages correctly?

dbarmaimon, 2019-04-23:
Hello guys, how is it going? I am brand new to ROS2 and I am facing the same issues trying to build the nav2_utils. I even got the master branch but maybe I am missing something.

What does it mean to delete the 'ros2_ws'? I tryied updating the repository from the master branch (the one in the link @routiful posted before)  with the VCS (after changing the permissions of the folder):
`sudo vcs import /opt/ros/crystal/ < ros2.repos `

And then  trying to build once again with:
`colcon build --symlink-install`

If anyone went forward, could you point me up how to follow. 

Thanks a lot.

P.D.:
The previous lines solved the issue with the nav_tasks (at least for me)
$ colcon build --packages-select bahaviortree_cpp
$ . ~/turtlebot3_ws/install/local_setup.bash
$ colcon build --symlink-install

corsaircpt, 2019-05-20:
hi, I am failing at the $ colcon build --packages-select-behaviortree_cpp stage . Any ideas would be appreciated, thank you

corsair@filika:~/projects/turtlebot3_ws$ colcon build --packages-select behaviortree_cpp
Starting >>> behaviortree_cpp
--- stderr: behaviortree_cpp                         
Traceback (most recent call last):
  File "/home/corsair/projects/ros2_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at /home/corsair/projects/ros2_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3
  /home/corsair/projects/ros2_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/corsair/projects/turtlebot3_ws/build/behaviortree_cpp/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  /home/corsair/projects/ros2_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:38 (include)
  /home/corsair/projects/ros2_ws/install/ament_cmake/share/ament_cmake/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  /home/corsair/projects/ros2_ws/install/ament_cmake/share/ament_cmake/cmake/ament_cmakeConfig.cmake:38 (include)
  CMakeLists.txt:44 (find_package)

---
Failed   <<< behaviortree_cpp	[ Exited with code 1 ]

Summary: 0 packages finished [3.72s]
  1 package failed: behaviortree_cpp
  1 package had stderr output: behaviortree_cpp

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

This issue will be solved due to navigation2 can be installed by binary.

closed 2019-08-21 by routiful

RIYAN-ROS, 2022-09-07:
$colcon build --symlink-install
Starting >>> turtlebot3_msgs
Starting >>> dynamixel_sdk
Starting >>> hls_lfcd_lds_driver
Starting >>> turtlebot3_description
Finished <<< turtlebot3_description [2.86s]                              
Starting >>> dynamixel_sdk_custom_interfaces
Finished <<< dynamixel_sdk [3.04s]                                       
Starting >>> turtlebot3_cartographer
Finished <<< turtlebot3_cartographer [0.73s]                             
Starting >>> turtlebot3_gazebo
Finished <<< turtlebot3_msgs [5.71s]                                      
Starting >>> turtlebot3_node
Finished <<< dynamixel_sdk_custom_interfaces [4.17s]                         
Starting >>> turtlebot3_navigation2
Finished <<< turtlebot3_gazebo [3.74s]                                       
Starting >>> turtlebot3_teleop
Finished <<< turtlebot3_navigation2 [0.84s]                                  
Starting >>> turtlebot3_example
Finished <<< turtlebot3_example [7.71s]                                        
Starting >>> turtlebot3_fake_node
Finished <<< turtlebot3_teleop [8.20s]                                         
Starting >>> ld08_driver
--- stderr: turtlebot3_fake_node                                               
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/src/turtlebot3_fake_node.cpp: In member function ‘void Turtlebot3Fake::init_parameters()’:
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/src/turtlebot3_fake_node.cpp:75:60: error: no matching function for call to ‘Turtlebot3Fake::declare_parameter<std::string>(const char [19])’
   75 |   this->declare_parameter<std::string>("joint_states_frame");
      |                                                            ^
In file included from /opt/ros/foxy/include/rclcpp/node.hpp:1224,
                 from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/include/turtlebot3_fake_node/turtlebot3_fake_node.hpp:21,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/src/turtlebot3_fake_node.cpp:17:
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note: candidate: ‘auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool) [with ParameterT = std::__cxx11::basic_string<char>; std::string = std::__cxx11::basic_string<char>; rcl_interfaces::msg::ParameterDescriptor = rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> >]’
  157 | Node::declare_parameter(
      | ^~~~
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note:   candidate expects 4 arguments, 1 provided
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/src/turtlebot3_fake_node.cpp:76:52: error: no matching function for call to ‘Turtlebot3Fake::declare_parameter<std::string>(const char [11])’
   76 |   this->declare_parameter<std::string>("odom_frame");
      |                                                    ^
n file included from /opt/ros/foxy/include/rclcpp/node.hpp:1224,
                 from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/include/turtlebot3_fake_node/turtlebot3_fake_node.hpp:21,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/src/turtlebot3_fake_node.cpp:17:
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note: candidate: ‘auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool) [with ParameterT = std::__cxx11::basic_string<char>; std::string = std::__cxx11::basic_string<char>; rcl_interfaces::msg::ParameterDescriptor = rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> >]’
  157 | Node::declare_parameter(
      | ^~~~
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note:   candidate expects 4 arguments, 1 provided
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/src/turtlebot3_fake_node.cpp:77:52: error: no matching function for call to ‘Turtlebot3Fake::declare_parameter<std::string>(const char [11])’
   77 |   this->declare_parameter<std::string>("base_frame");
      |                                                    ^
In file included from /opt/ros/foxy/include/rclcpp/node.hpp:1224,
                 from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/include/turtlebot3_fake_node/turtlebot3_fake_node.hpp:21,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_fake_node/src/turtlebot3_fake_node.cpp:17:
[... cut ...]
Aborted  <<< turtlebot3_node [26.9s]               
Aborted  <<< ld08_driver [16.9s]
Aborted  <<< hls_lfcd_lds_driver [39.2s]                                   
                                   
Summary: 9 packages finished [39.7s]
  1 package failed: turtlebot3_fake_node
  3 packages aborted: hls_lfcd_lds_driver ld08_driver turtlebot3_node
  3 packages had stderr output: hls_lfcd_lds_driver turtlebot3_fake_node turtlebot3_node
  4 packages not processed

i have face this kind of error please to tell the solution