ROBOTIS-GIT/turtlebot3 #403: USB Power OK, but doesnt turn on with 11.1V battery or 12V SMPS power supply

rahulrajd, 2019-04-15:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ *] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [* ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ *] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ *] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

  If i power the opencr board with USB power, the green light turns on. But the board is not responding for any 11.1V power supply via battery or a 12v power supply. Do you think the fuse has been blown out ?

OpusK, 2019-04-16:
Hi, @rahulrajd 

Would you like to check if the power switch is ON?
And, do you mean that even if you connect the power, the green LED is off?

rahulrajd, 2019-04-16:
Yes. No LEDs turn on even when the switch is on. 
I verified the power supply via digital multimeter, seems like fuse and the power inputs are okay

OpusK, 2019-06-12:
@rahulrajd 

Sorry for the delay.

> Yes. No LEDs turn on even when the switch is on.

I think it's a hardware problem.
It may be a regulator problem.
If you can check it yourself, would you check if the LED goes on normally?

If you want hardware repair, please contact `support@robotis.com` or leave a message at [ROBOTIS Forum](http://en.robotis.com/service/forum.php).

rjshim, 2019-10-15:
As there have been no updates, I am closing this issue.

closed 2019-10-15 by rjshim