ROBOTIS-GIT/turtlebot3 #405: Turtlebot 3 navigation: erroneous movement

master-chrise, 2019-04-18:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [X] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - DWA planner failed to produce path.
  
8. Please, describe detailedly what difficulty you are in 

First of all the TB3 (waffle pi) navigation works well using the following parameter set:

     https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_navigation/param

However there are sometimes errors in movement direction of the TB3 (real or simulated).
Best to see on point to point target movement in the same room axis. 
On error TB3 doesn't change its direction if you change the targets when TB3 is still running (however RViz displays the correct path) . So TB3 runs (wrong path) to the next wall and then starts the correct path. Wall "contact" will cause the error massage "DWA planner failed to produce path.".

RyanJaehyunShim, 2019-04-19:
@master-chrise 

Would you test below and let us know if you are having the same issue? 
$ roscore
$ roslaunch turtlebot3_gazebo turtlebot3_world.launch
$ roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml  
(map.yaml file is in the turtlebot3_navigation package.)

Also, if you need help with your own work about our tutorial, can you let us know which tutorial (with what terminal commands) you tried & send error codes plus the screenshot of whatever that could help us figure out what issue you are having.  

Regards,
Ryan

master-chrise, 2019-04-19:
@RyanJaehyunShim

Thank you for your prompt reply.

Command history:
$ roscore
$ roslaunch turtlebot3_gazebo turtlebot3_house.launch
$ roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml

The following pictures illustrate the problem:

**- Set a new goal (green arrow):**
[image]

- **TB3 doesn't navigate to the new goal**:
[image]
[image]

- **Shortly before TB3 touches the wall, TB3 starts to follow the correct path** 
[image]
[image]
[image]

It looks like the problem has occurred since tag version 1.0.0 (see below)

https://github.com/ROBOTIS-GIT/turtlebot3/tree/1.0.0/turtlebot3_navigation/param

Previous versions work correctly...

Regards,
master-chrise

RyanJaehyunShim, 2019-04-22:
@master-chrise 

I think the problem occurred because of incorrect parameter setting of the robot. Please refer to the link below and modify the related files in /.../turtlebot3_navigation/param folder.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#tuning-guide)

Let us know if it doesn't work,
Ryan

master-chrise, 2019-04-23:
@RyanJaehyunShim 

It looks like the problems are caused by the forward simulation parameter "controller_frequency" (default: 10 Hz, dwa_local_planner_params_waffle_pi.yaml). Setting the parameter to a lower value (e.g. 4 Hz) solves my problem.
However, I'm currently not sure regarding the difference meaning of the parameters "controller_frequency" in the "dwa_local_planner_params_waffle_pi.yaml" and "move_base_params.yaml" file...

Regards,
master-chrise

RyanJaehyunShim, 2019-04-24:
@master-chrise 

You will only need to change parameters in the dwa_local_planner_params_waffle_pi.yaml file when it comes to navigation path planning. The controller frequency in the move_base_params.yaml is for the entire move_base controller (please refer to the following link if you have more questions about move_base. http://wiki.ros.org/move_base)

Also, the problem you mentioned regarding a turtlebot moving backward instead of following the path to a new goal can be solved by setting 'min_vel_x' value to zero.

Hope it helps you,
Ryan

RyanJaehyunShim, 2019-05-16:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if you have any other questions. Thanks.

closed 2019-05-16 by RyanJaehyunShim