ROBOTIS-GIT/turtlebot3 #406: How can I move my robot along a straight line with odometry and get the position

zl390, 2019-04-23:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

My problem is that I want my turtlebot to move along a straight line for 
a certain distance and get the position of my tuetlebot.

So I have tried the turtlebot3_pointop_key to move my turtlebot and 
print the position. But I find that it does not move along a straight 
line and not give the correct position due to the effect of the ground. 

And when I use navigation, I find that odometry might be able to give a more accurate position.

Hence I would like to ask how I can use odometry to give a feedback so 
that my turtlebot can move along a straight line and how I can store the 
position from odometry.

RyanJaehyunShim, 2019-04-24:
@zl390 

I think the problem occurred because of wheel slip. To reduce the influence of it, you can make it heavier (by putting a weight on the robot). You can also add sensors and cameras to your robot and implement feedback control system on your own (unfortunately, we do not support each one's own work).

Also, if you would like to know how odometry data are used in localization, please refer to the following link.
(http://ais.informatik.uni-freiburg.de/teaching/ws17/mapping/pdf/slam03-bayes-filter-short.pdf)

Regards,
Ryan

kijongGil, 2019-04-24:
Hi @zl390 
You can use the odometry by referring to [here](http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom).
However, using only odometry can not control the robot to move in a straight line. Because the odometry is only informing the position of the robot caused by the rotation of the motor. So, when robot has wheel slip, It has nothing to do with the odometry.
I think you need more sensor as IMU to control the robot to move in a straight line.

Thanks,
Gilbert

zl390, 2019-04-24:
Thank you for your help. 
If the odometry cannot deal with wheel slip, is it any way that I could get more accurate position of the robot? And where can I find more about IMU sensor?

Kind regards
Zheng

kijongGil, 2019-04-25:
Turtlebot3 has already IMU sensor on OpenCR. You can check IMU topic following command after bringup robot.

`rostopic echo /imu`

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman