ROBOTIS-GIT/turtlebot3 #407: Pi camera on Turtlebot3

umarfarooqumzen, 2019-04-23:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [tick] Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ tick] Raspberry Pi 3
3. Which OS you installed in SBC?

    - [tick] Ubuntu MATE 16.04 or later

4. Which OS you installed in Remote PC?

    - [ tick] Ubuntu 16.04 LTS (Xenial Xerus)  
8. Please, describe detailedly what difficulty you are in 
I have mounted pi camera v2 on turtlebot 3 burger. and i want to develop a local website on which i can show camera feed and control the robot. i have the code for the website, control to move are working but i am not able to show LDS results or pi camera feed into my local web. Can anyone please help ASAP. i am in mid of a university project.

thanks in advance.

RyanJaehyunShim, 2019-04-24:
@umarfarooqumzen 

As to what you mentioned you need help with, please refer to the following website (http://robotwebtools.org/tools.html).

You might also want to refer to the following link to know which topics you will need to use. 
(http://emanual.robotis.com/docs/en/platform/turtlebot3/basic_operation/#basic-operation
http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#tutorials-4-calibration)

Regards,
Ryan

umarfarooqumzen, 2019-04-24:
Believe me i tried all these but i am not able to get feed from the camera to the web tools.i used the same resoirces as you replied me with. ROS control mobile app is showing both LDS data and camera feed by the use of "image_raw/compressed" but on web it is not working the rqt_graph shows that the pi camera is attached to the android app but not to the web. Plz can someone send me exact code for the turtlebot3 burger with pi camera so that i can control it using web tool.

If you have any code or resource plz help me ASAP

RyanJaehyunShim, 2019-04-25:
Your program will need to subscribe to the topic you mentioned and the following link is how to write a subscriber (http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29).

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman