ROBOTIS-GIT/turtlebot3 #408: OpenCR left and right DYNAMIXELs are moving in opposite direction

MHammadA, 2019-04-25:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - > Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - > Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - > Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - > Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

Hello everyone
I am new with turtlebot3.  I am following the official manual for turtlebot3 http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/. When I upload the firmware using shell script for OpenCR, and after firmware successfully uploaded. I tried to test the OpenCR and Dynamixels by following the instruction given at the end. I pushed the sw 1, dynamixels are working but instead of moving forward it moves in rotating for both sw 1 and sw 2, i updated the firmware again but the result is same.
I don't know what the issue is i searched for it but i didn't find any solution. Plz help me if anybody have any idea about this issue.
Thanks in advance.

ROBOTIS-Kayman, 2019-04-29:
Hi,

I think the problem is about Dynamixels.
Could you follow [this instructions](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3)?

Regards,
Kayman

MHammadA, 2019-04-29:
Thank you so much for response.
Yeah the problem was with Dynamixels. So i followed the instruction in the given link, now the problem is solved.
Thanks Kayman

closed 2019-04-29 by MHammadA