ROBOTIS-GIT/turtlebot3 #409: ROS2 - Turtlebot3 odometry is approx 0.5x what it should be

mkhansen-intel, 2019-04-26:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [x ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ x] Intel NUC

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [x ] Ubuntu 18.04

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: ROS2 latest
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - Followed the ROS2 instructions here:
http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2
 
7. Copy and Paste your error message on terminal

    - No errors
  
8. Please, describe detailedly what difficulty you are in 

    - We have seen that the robot odom value is approximately 0.5x of the actual distance the robot travels when operating with teleop keyboard.

We measured this 3 times with different distances and velocities, and it appears that the robot:
1. Publishes values of distance that is 0.5x
2. Publishes values of velocity that is 0.5x. 

We tried to isolate where this is occuring in the code but we haven't so far. We DON'T see this problem in simulation, so maybe that is a clue that it is something in the firmware, but we don't know.

Please test with ROS2 on a robot and verify odom values are correct, or see if you can find the bug. We'd really like to do a demo with TB3 but this is causing SLAM and Navigation to be very confused.

routiful, 2019-04-29:
@mkhansen-intel 
Can you describe more detail how to test it? Inaccurate odometry may be caused by condition of floors or contact problem btw wheels and floors due to light robot weight.

mkhansen-intel, 2019-04-30:
I found the issue to be uninitialized variables and filed PR #411 to fix. Please take a look.

routiful, 2019-05-02:
I merged that PR.

closed 2019-06-10 by mkhansen-intel