ROBOTIS-GIT/turtlebot3 #410: catkin_make failed (turtlebot3)

Aidlieys, 2019-04-29:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [o ] Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3

3. Which OS you installed in SBC?

    - [ ] Raspbian

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

 
6. Write down the commands you used in order

    - 
1.
    $ cd ~/catkin_ws/src
    $ git clone https://github.com/ROBOTIS-GIT/turtlebot3_autorace.git 

2.
    $ git clone https://github.com/arixrobotics/robosot_race.git

3.
    $ git clone https://github.com/arixrobotics/robosot_race_gazebo.git

4. 
    $ cd ~/catkin_ws
    $ catkin_make
 
7. Copy and Paste your error message on terminal

- 
does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- +++ processing catkin package: 'turtlebot3_description'
-- ==> add_subdirectory(turtlebot3/turtlebot3_description)
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:116 (add_subdirectory):
  The source directory

    /home/pi/catkin_ws/src/turtlebot3/turtlebot3_description

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:990: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
  
8. Please, describe detailedly what difficulty you are in 

    - Hi, This will not be the problem in my laptop but just in my turtlebot. I'm posting a question, hoping that there might be someone who experienced same problem during installing the turtlebot. i hope someone can help me to solve this problem or Error ^-^.. If anyone have any suggestions, please do let me know.

RyanJaehyunShim, 2019-04-30:
@Aidlieys 

The packages you are having the issue with are not need in Turtlebot PC.

$ cd ~/catkin_ws/src/turtlebot3
$ sudo rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/

Regards,
Ryan

Aidlieys, 2019-04-30:
> @Aidlieys
> 
> The packages you are having the issue with are not needed in Turtlebot PC.
> 
> $ cd ~/catkin_ws/src/turtlebot3
> $ sudo rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/
> 
> Regards,
> Ryan

So i need to put this program in PC or Turtlebot??

RyanJaehyunShim, 2019-04-30:
@Aidlieys 

You should do it in your Tuetlebot. 

As you can see on our e-manual, Turtlebot PC indicates the SBC in your Turtlebot, whereas Remote PC means your own PC (like your laptop).

Let us know if you have any other questions, 
Ryan

Aidlieys, 2019-04-30:
> @Aidlieys
> 
> You should do it on your Tuetlebot.
> 
> As you can see on our e-manual, Turtlebot PC indicates the SBC in your Turtlebot, whereas Remote PC means your own PC (like your laptop).
> 
> Let us know if you have any other questions,
> Ryan

i have done do at my PC. and it become error when i try to doing turtlebot_slam.. how to fixed it back??

the command out :
> slam_methods:=gmapping
[turtlebot3_slam.launch] is neither a launch file in package [turtlebot3_slam] nor is [turtlebot3_slam] a launch file name
The traceback for the exception was written to the log file

RyanJaehyunShim, 2019-04-30:
@Aidlieys 

Please, check out our e-manual.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup)

Regards,
Ryan

Aidlieys, 2019-04-30:
> @Aidlieys
> 
> Please, check out our e-manual.
> (http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup)
> 
> Regards,
> Ryan

can help me solve another problem?? hehe ^-^

i try to bringup the camera..

 $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_camera_pi.launch

command out :

[turtlebot3_autorace_camera_pi.launch] is neither a launch file in package [turtlebot3_autorace_camera] nor is [turtlebot3_autorace_camera] a launch file name
The traceback for the exception was written to the log file

RyanJaehyunShim, 2019-04-30:
@Aidlieys 

Please refer to the following link.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#tutorials-1-requirements)

Also, as we cannot support every little issue you encounter while following the tutorial, I recommend you learn basic knowledge first at the following link (http://wiki.ros.org/ROS/Tutorials) and follow our e-manual. 

Regards,
Ryan

Aidlieys, 2019-04-30:
Ok.. Thank for Help..

RyanJaehyunShim, 2019-04-30:
@Aidlieys 

I will close this issue. Good luck with your work! Ryan

closed 2019-04-30 by RyanJaehyunShim