ROBOTIS-GIT/turtlebot3 #412: Wheel encoder raw data processing

Fugashu, 2019-04-30:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)
  
5. Please, describe detailedly what difficulty you are in 

Hello,

I want to use an EKF and an UKF for localization purposes. 
 
Can you provide me with the source code where the raw data of the wheel encoders is processed and used for the Odometry?

I appreciate your help.

RyanJaehyunShim, 2019-05-02:
@Fugashu 

Please check out the following link and path. 
(http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup)
/OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino

Regards,
Ryan

Fugashu, 2019-05-07:
Thank you, I found what I needed in the specified file!

closed 2019-05-07 by Fugashu