ROBOTIS-GIT/turtlebot3 #413: Scan topic publishing at 30hz, scan_half only at 5hz

mkhansen-intel, 2019-04-30:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ x] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ x] etc (Intel NUC)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ x] etc (Ubuntu 18.04)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

I noticed when running TB3 that the /scan topic runs at 30hz, while the /scan_half is only at 5hz. I would expect that since /scan_half is converted to /scan, the frequencies should be the same. 

This isn't causing a functional error, but seems like it will be re-publishing the same data on the /scan topic multiple times before new data arrives on the /scan_half. The nodes processing this data are therefore getting callbacks to process the same data over and over again. This is a waste of power.

Please correct me if I have something wrong here. Otherwise, either the lidar should be running at 30hz or the /scan topic should be slowed down to 5hz.

routiful, 2019-05-02:
Because some problem of RTPS, All scan data(every 1 deg)has not been published yet.
So I took some trick to get 360 deg every 2 deg.

You may check the code to understand that.

Publisher(Scan half)
https://github.com/ROBOTIS-GIT/turtlebot3/blob/a4039869a8cb75b0303c88ca790ebb1ddb1c89d9/turtlebot3_lidar/turtlebot3_lidar/main.cpp#L167

Publisher(Scan)
https://github.com/ROBOTIS-GIT/turtlebot3/blob/a4039869a8cb75b0303c88ca790ebb1ddb1c89d9/turtlebot3_node/src/lidar.cpp#L38

mkhansen-intel, 2019-05-02:
That's fine but I'm just talking about the frequency mismatch between the two topics. /scan_half runs at 5hz, and /scan runs at 30hz. If the /scan topic is integrating the /scan_half data, then it should run at the same frequency, right?

routiful, 2019-05-03:
@mkhansen-intel Your right. It should be. 
@OpusK Can you check that the /scan_half get better to work like /scan ?

mkhansen-intel, 2019-05-06:
According to the LDS spec, it should be capable of running faster than 5hz, I believe it says 300+/- 10.
http://emanual.robotis.com/assets/docs/LDS_Basic_Specification.pdf

If it can be made to run at 30hz that would match the scan topic frequency.

routiful, 2019-05-07:
@mkhansen-intel 
I've modified frequency from 5 to 30. (https://github.com/ROBOTIS-GIT/turtlebot3/commit/81708d6b09a7492912835e6a54a44292cf954721)
Please pull new commits and re-build `Lidar Client`
http://emanual.robotis.com/docs/en/popup/turtlebot3_ros2_sbc_setting

mkhansen-intel, 2019-05-07:
OK, I read the spec wrong. The lidar says "300 +/- 10 rpm" which is revolutions per MINUTE. I was thinking it was per SECOND.

In that case, 5hz is the best it can do and instead of changing them both (/half_scan and /scan) to 30 hz, they should both be 5 hz.

mkhansen-intel, 2019-05-07:
So, I believe you need to revert the change you made and instead change this:
https://github.com/ROBOTIS-GIT/turtlebot3/blob/f0fa3e70fc51fdc93b38f6ce3389de8ab9c29167/turtlebot3_node/src/constants.h#L33

change 33ms to 200ms

Then they will both be matched and running at 5hz.

Sorry for the confusion!

routiful, 2019-05-08:
Sorry, I didn't check it before. It is correct to set 5hz both topics.

closed 2019-05-08 by mkhansen-intel