ROBOTIS-GIT/turtlebot3 #416: [bug][turtlebot3_slam] init() not called if the macro NDEBUG was defined

developer-yj, 2019-05-07:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [*] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [*] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [*] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [*] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)
  
5. as title said, `ROS_ASSERT(init());` init() not called if the macro NDEBUG was defined

RyanJaehyunShim, 2019-05-07:
@developer-yj 

Would you explain the problem in more detail? Describing what modifications you made & error messages will help us figure out a solution to it.

Regards,
Ryan

developer-yj, 2019-05-07:
On ubuntu 16.04, x64, installed by command: sudo apt-get install ros-kinetic-turtlebot3-slam，test commands:
# terminal0
$ roscore

# terminal1
$ rosrun turtlebot3_slam flat_world_imu_node

# terminal2
rosnode info /flat_world_imu_node 
--------------------------------------------------------------------------------
Node [/flat_world_imu_node]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: None

Services: 
 * /flat_world_imu_node/get_loggers
 * /flat_world_imu_node/set_logger_level

contacting node http://ubuntu:44317/ ...
Pid: 4066
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS

You can see, /flat_world_imu_node not subscribed or published any topic, because  `init()` not called.

fix it:
====
ROS_ASSERT(init()); 
====>
bool ret = init();
ROS_ASSERT(ret);
====

RyanJaehyunShim, 2019-05-07:
@developer-yj 

Could you please which tutorial # you are working on now? DId you bring up the Turtlebot with below?
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch

Regards,
Ryan

ROBOTIS-Kayman, 2019-05-09:
@developer-yj 

Thank you for your contribution.
We will fix it :)

Regards, 
Kayman

closed 2019-07-19 by ROBOTIS-Kayman