ROBOTIS-GIT/turtlebot3 #420: Fix goals in the map using SLAM and navigation

MHammadA, 2019-05-09:
Hi 
I am working with Turtlebot3 waffle pi. I am using SLAM and navigation example form the Turtlebot3 manual. I have a question that i wanted to specify or fix the goal on the map created by SLAM. What i mean is i don't want to use the RViz to specify the goal. I just want to fix the goal in the map so that it can automatically go to the goal in the map instead of using visualization tool. 
So is it possible?

RyanJaehyunShim, 2019-05-09:
@MHammadA 

Please refer to the link below. 
(http://wiki.ros.org/navigation/Tutorials/SendingSimpleGoals)

If you would like to set the initial position of the robot on the map, refer to the following file.
~/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo/launch/turtlebot3_world.launch 

Regards,
Ryan

MHammadA, 2019-05-09:
@RyanJaehyunShim 

Thanks you so much for responding.
I actually follow the following tutorial.
https://hotblackrobotics.github.io/en/blog/2018/01/29/seq-goals-py/
It is working but i want to use it for real turtlebot not using gazebo. Sorry i am beginner with turtlebot so i don't know exactly how to change it, and i have very minimum time. So please if you can help me to explain a little bit.

Thanks in advance.

RyanJaehyunShim, 2019-05-10:
@MHammadA 

I do not understand your question clearly. Are you asking how to use an acutal physical robot? If yes, refer to the link below. There would be no difference in how you use the package.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup)

Or if you are asking how to set the initial position of the robot, it is very simple, just put it in your map. 

Regards,
Ryan

MHammadA, 2019-05-13:
@RyanJaehyunShim 

I have solved this issue by applying some changes in the launch files now the code is working.
Thankyou for your help.

Regards,
Hammad

closed 2019-05-13 by MHammadA