ROBOTIS-GIT/turtlebot3 #421: http://emanual.robotis.com/docs/en/platform/turtlebot3/basic_examples/#patrol

liusanchuan, 2019-05-09:
i find the demo  [patrol](http://emanual.robotis.com/docs/en/platform/turtlebot3/basic_examples/#patrol) is not very precision, So i want to combine with move_base to patrol, but i don‘t know how to send the my costumed trajectory path to move_base。

RyanJaehyunShim, 2019-05-10:
@liusanchuan 

If you would like to know how to use move_base, please refer to the link below.
(http://wiki.ros.org/move_base)

Also, the following link might give you a clue about how you should make modifications to your program.
(https://github.com/ROBOTIS-GIT/turtlebot3/issues/405)

Regards,
Ryan

ROBOTIS-Kayman, 2019-07-19:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-07-19 by ROBOTIS-Kayman