ROBOTIS-GIT/turtlebot3 #422: Navigation with TB3 Waffle Pi - Parameter Tuning

roboqu, 2019-05-10:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ x ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

8. Please, describe detailedly what difficulty you are in 

I tested your navigation tutorial for the TurtleBot3 Waffle Pi. When trying to tune the parameters mentioned on http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#tuning-guide, I tested different inflation_radius and cost_scaling_factor values, but in the cost map shown in Rviz nothing changes. The ROS parameters shown when starting the node were updated as specified in the parameter.yaml file. 
But according to the screenshots shown on your webpage, the region marked with blue should increase/decrease when changing the inflation_radius and the red region with the cost_scaling_factor. 
Can you help me to find the problem, please?

RyanJaehyunShim, 2019-05-14:
@roboqu 

Would you check the following three points?
1. modified $(model) properly
2. kill and rerun $ roscore after modifying the yaml file
3. check the parameters (i.e, inflation_radius, cost_scaling_factor) using $ rqt
(http://wiki.ros.org/rqt_reconfigure)

Regards,
Ryan

roboqu, 2019-05-14:
Thanks so much! Killing and rerunning the roscore after modifying the yaml file did the magic :)

closed 2019-05-14 by roboqu