ROBOTIS-GIT/turtlebot3 #423: Turtlebot3 + ROS Melodic + Ubuntu 18.04 for remote computer

pitosalas, 2019-05-13:
Is there any known reason to stick to Ubuntu 16.04 and ROS Kinetic as the emanual says?
Or is Ubuntu 16.04 (on remote) + ROS Melodic fully supported for the TB3 software provided here? 

What is the recommendation?

Thanks!

ROBOTIS-Kayman, 2019-05-14:
Hi,

TB3 fully support ROS melodic, but some dependency packages do not. So If you use ROS melodic, you have to manually compile these packages. (ex. gmapping)

check the following list.
https://index.ros.org/packages/#melodic

Regards, 
Kayman

pitosalas, 2019-05-15:
Thanks. Can you give (or link to) instructions on how to “correctly” manually compile the packages? Thanks,

Pito Salas
Brandeis Computer Science
Volen 134

ROBOTIS-Kayman, 2019-05-20:
This is ros wiki for building the packages : [LINK](http://wiki.ros.org/ROS/Tutorials/BuildingPackages)

1. Download the source to the catkin workspace
  The location of the source file can be found on the [ros wiki](http://wiki.ros.org/).
2. Build a package using `catkin_make`

In case of gmapping, it may not work in melodic due to the development stopped in the hydro version.

Regards,
Kayman

yoshimalucky, 2019-05-28:
@pitosalas What is your reason of choosing Melodic and 18.04?

pitosalas, 2019-05-28:
I would like to move to 18.04. I am trying to determine if I should do that. Kinetic is 16.04 only, so to go to 18.04 I need to use melodic.

Pito Salas
Brandeis Computer Science
Volen 134

ROBOTIS-Kayman, 2019-07-19:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-07-19 by ROBOTIS-Kayman

amna786-rajpoot, 2019-09-17:
can u give me the link to install packages of turtlebot3 on ros melodic??

rjshim, 2019-09-18:
@amna786-rajpoot 

There is no instruction for TB3 Melodic but you can refer to [this link](http://emanual.robotis.com/docs/en/platform/openmanipulator_pro/ros_setup/#install-ubuntu-on-pc). You can see the following three points that you need to note.

1. Install Ubuntu 18.04 instead of Ubuntu 16.04
2. Install ROS Melodic instead of Kinetic (you can download a shell script file for melodic install from our eManual by using wget)
3. Install relevant packages for melodic instead of kinetic 
(e.g., 'sudo apt-get install ros-melodic-ros-controllers' instead of 'sudo apt-get install ros-kinetic-ros-controllers') 
If there is no existing binary file, you will need to find source file for it and catkin_make with other packages as mentioned in the early conversation by pitosalas and Kayman in this thread. 

Let us know if you have any other questions, 
Ryan

kjyothiswaroop, 2020-08-10:
I have tried doing that I have installed all the packages in Melodic but somehow TB3 doesn't seem to work on Melodic. In the coming future will TB3 be made compatible with ROS Melodic?

RahulSTrivedi, 2020-08-10:
@kjyothiswaroop I have the same problem. Turtlebot3 doesn't work with Ubuntu 18 and ros melodic. What should we do? Can anyone help?

rjshim, 2020-08-11:
@kjyothiswaroop @RahulSTrivedi 

Please open a new issue for the question. 

Thanks, 
Ryan