ROBOTIS-GIT/turtlebot3 #426: Update turtlebot3 to sync ROS2 Dashing

routiful, 2019-05-16:
The update will be published after release Dashing(stable version)
The purpose of this issue is to inform updates and to get some opinion to ROS users which functions need to be included.

mkhansen-intel, 2019-06-10:
Do you have a status update on this? I'm trying to test Navigation2 with TB3 on Dashing and running into TF issues. I'll see if I can figure out what's going on but wanted to check first if you're working on fixes for Dashing.

routiful, 2019-06-11:
@mkhansen-intel 
I will inform you until this week :)

testkit, 2019-06-21:
@routiful , just for your information, eProsima/Micro-XRCE-DDS-Agent#90 was merged several days ago.

mkhansen-intel, 2019-06-28:
My issues were not related to TB3 code and we have been using TB3 in simulation in preparation for our Dashing release very soon. What's your status for Dashing?

routiful, 2019-07-01:
@mkhansen-intel 
I have done to update the code for Dashing including teleop, cartographer in Gazebo.
You can pull the code at ros2 branch.

But I got some tf error when I tested Navigation2.
If you have any clue about it, please comment for me.
I've attached execute cmd and log msg below

```
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
$ ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True
```

```
[INFO] [launch]: All log files can be found below /home/darby/.ros/log/2019-07-01-14-34-51-894242-kingod-21870
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [21905]
[INFO] [map_server-2]: process started with pid [21906]
[INFO] [amcl-3]: process started with pid [21907]
[INFO] [world_model-4]: process started with pid [21908]
[INFO] [dwb_controller-5]: process started with pid [21909]
[INFO] [navfn_planner-6]: process started with pid [21910]
[INFO] [motion_primitives_node-7]: process started with pid [21911]
[INFO] [bt_navigator-8]: process started with pid [21918]
[INFO] [rviz2-9]: process started with pid [21919]
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Starting the system bringup...
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
[map_server-2] [INFO] [map_server]: Creating
[world_model-4] [INFO] [world_model]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating map_server
[dwb_controller-5] [INFO] [dwb_controller]: Creating
[motion_primitives_node-7] [INFO] [motion_primitives]: Configuring Spin
[motion_primitives_node-7] [INFO] [motion_primitives]: Robot: Creating
[bt_navigator-8] [INFO] [bt_navigator]: Creating
[amcl-3] [INFO] [amcl]: Creating
[motion_primitives_node-7] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-9] qt5ct: using qt5ct plugin
[dwb_controller-5] [INFO] [local_costmap.local_costmap]: Creating
[motion_primitives_node-7] [INFO] [motion_primitives]: Configuring BackUp
[motion_primitives_node-7] [INFO] [motion_primitives]: Robot: Creating
[navfn_planner-6] [INFO] [navfn_planner]: Creating
[world_model-4] [INFO] [global_costmap.global_costmap]: Creating
[rviz2-9] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-9] [INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-9] qt5ct: D-Bus global menu: no
[rviz2-9] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[map_server-2] [INFO] [map_server]: Configuring
[map_server-2] [INFO] [map_server]: OccGridLoader: Creating
[map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
[map_server-2] [WARN] [map_server]: Mode parameter not set, using default value (trinary)
[map_server-2] [INFO] [map_server]: Activating
[map_server-2] [INFO] [map_server]: OccGridLoader: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating amcl
[amcl-3] [INFO] [amcl]: Configuring
[amcl-3] [INFO] [amcl]: initTransforms
[amcl-3] [INFO] [amcl]: initPubSub
[amcl-3] [INFO] [amcl]: Subscribed to map topic.
[amcl-3] [INFO] [amcl]: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating world_model
[world_model-4] [INFO] [world_model]: Configuring
[world_model-4] [INFO] [global_costmap.global_costmap]: Configuring
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "static_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: StaticLayer: Requesting map from the map service
[map_server-2] [INFO] [map_server]: OccGridLoader: Handling map request
[world_model-4] [INFO] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 384 X 384 at 0.050000 m/pix
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
[world_model-4] [INFO] [global_costmap.global_costmap]: Subscribed to Topics: scan
[world_model-4] [INFO] [global_costmap.global_costmap]: Using plugin "inflation_layer"
[world_model-4] [INFO] [world_model]: Activating
[world_model-4] [INFO] [global_costmap.global_costmap]: Activating
[world_model-4] [INFO] [global_costmap.global_costmap]: Checking transform
[rviz2-9] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-9] Parsing robot urdf xml string.
[rviz2-9] [ERROR] [rviz2]: "map" passed to lookupTransform argument target_frame does not exist. 
[rviz2-9] [ERROR] [rviz2]: "map" passed to lookupTransform argument target_frame does not exist. 
[rviz2-9] [ERROR] [rviz2]: "map" passed to lookupTransform argument target_frame does not exist. 
[rviz2-9] [ERROR] [rviz2]: "map" passed to lookupTransform argument target_frame does not exist. 
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.002 timeout was 1.
[... cut ...] c/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[rviz2-9] [ERROR] [rviz2]: Failed to create 2 swatches_
[rviz2-9] [INFO] [rviz2]: Creating 4 swatches_
[rviz2-9] [ERROR] [rviz2]: "map" passed to lookupTransform argument target_frame does not exist. 
[amcl-3] [WARN] [amcl]: Waiting for map....
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.099 timeout was 1.
```

mkhansen-intel, 2019-07-01:
@routiful - do you have a branch where you've made changes for Dashing? If so I can pull it and try this. I tried it with your default branch and noticed a couple things, you may have resolved them already:
1. You can't use the 'nav2_bringup_1st_launch.py' and 'nav2_bringup_2nd_launch.py' files anymore. You should be able to simply use 'nav2_bringup_launch.py' instead.
2. I believe you're using a different parameters file than we are, if so, you'll need to incorporate the changes we made to nav2_params.yaml in the nav2_bringup/launch folder. It looks like you may have done that because I can see in your message above you have the lifecycle_manager node running and starting all the nodes for you, this is good. 
3. Have you checked to make sure the /map topic is being published? Does it appear in RVIZ? What if you do 'ros2 topic hz /map'? Do you see it being published every 2 seconds?

routiful, 2019-07-02:
@mkhansen-intel 
I've updated the code into ros2 branch (https://github.com/ROBOTIS-GIT/turtlebot3/commit/3d875af21e68f13f1161282a77328aa4a9e4cdf1)
ros2 branch will be pushed to dashing-devel branch.

1. I've tested `nav2_bringup_launch.py` with new parameters. 
https://github.com/ROBOTIS-GIT/turtlebot3/blob/3d875af21e68f13f1161282a77328aa4a9e4cdf1/turtlebot3_navigation2/launch/navigation2.launch.py#L60
https://github.com/ROBOTIS-GIT/turtlebot3/blob/3d875af21e68f13f1161282a77328aa4a9e4cdf1/turtlebot3_navigation2/param/burger_params.yaml#L1

1. I've checked the /map topic using `ros2 topic hz /map` . It is published every 2 secs.
```
darby@kingod:~$ ros2 topic hz /map
[INFO] [rmw_coredds_cpp]: RMW successfully initialized with CoreDDS
average rate: 0.500
	min: 2.001s max: 2.002s std dev: 0.00016s window: 2
average rate: 0.500
	min: 1.998s max: 2.002s std dev: 0.00148s window: 3
average rate: 0.501
	min: 1.990s max: 2.002s std dev: 0.00469s window: 4
average rate: 0.500
	min: 1.990s max: 2.009s std dev: 0.00607s window: 5
average rate: 0.500
	min: 1.990s max: 2.009s std dev: 0.00574s window: 6
average rate: 0.500
	min: 1.990s max: 2.009s std dev: 0.00549s window: 7
average rate: 0.500
	min: 1.990s max: 2.009s std dev: 0.00514s window: 8
average rate: 0.500
	min: 1.990s max: 2.009s std dev: 0.00485s window: 9
average rate: 0.500
	min: 1.990s max: 2.009s std dev: 0.00463s window: 10
average rate: 0.500
	min: 1.988s max: 2.009s std dev: 0.00565s window: 11
average rate: 0.500
	min: 1.988s max: 2.015s std dev: 0.00692s window: 12
average rate: 0.500
	min: 1.988s max: 2.015s std dev: 0.00673s window: 13
average rate: 0.500
	min: 1.988s max: 2.015s std dev: 0.00650s window: 14
```

1. I've attached screenshot when I execute nav2 launch file

[image]

routiful, 2019-07-04:
Connected issue https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/99

mkhansen-intel, 2019-07-22:
@routiful - thanks for the updates, I commented on the other thread you linked also

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

This issue was solved to apply new firmware.

closed 2019-08-21 by routiful