ROBOTIS-GIT/turtlebot3 #427: ''ImportError" during Follower Demo

ColinJou, 2019-05-17:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: Latest (downloaded 16/05/19)
    - Firmware version: Latest (downloaded 16/05/19)
 
6. Write down the commands you used in order

    - PC : roscore
    - Turtlebot : roslaunch turtlebot3_bringup turtlebot3_robot.launch
    - PC : roslaunch turtlebot3_bringup turtlebot3_remote.launch
    - PC : roslaunch turtlebot3_follow_filter turtlebot3_follow_filter.launch
    - PC : roslaunch turtlebot3_follower turtlebot3_follower.launch
 
7. Copy and Paste your error message on terminal

    process[follower-1]: started with pid [4686]
[INFO] [1558107315.566191]: Follower node initialized
Traceback (most recent call last):
  File "/home/student/catkin_ws/src/turtlebot3_applications/turtlebot3_follower/nodes/follower", line 137, in <module>
    main()
  File "/home/student/catkin_ws/src/turtlebot3_applications/turtlebot3_follower/nodes/follower", line 132, in main
    follow = follower()
  File "/home/student/catkin_ws/src/turtlebot3_applications/turtlebot3_follower/nodes/follower", line 31, in __init__
    self.clf = pickle.load(open(self.config_dir + '/clf', "rb"))
  File "/usr/lib/python2.7/pickle.py", line 1384, in load
    return Unpickler(file).load()
  File "/usr/lib/python2.7/pickle.py", line 864, in load
    dispatch[key](self)
  File "/usr/lib/python2.7/pickle.py", line 1096, in load_global
    klass = self.find_class(module, name)
  File "/usr/lib/python2.7/pickle.py", line 1130, in find_class
    __import__(module)
ImportError: No module named sklearn.ensemble.forest
[follower-1] process has died [pid 4686, exit code 1, cmd /home/student/catkin_ws/src/turtlebot3_applications/turtlebot3_follower/nodes/follower __name:=follower __log:=/home/student/.ros/log/2e06e492-78b9-11e9-b0bd-000c293d2fe3/follower-1.log].
log file: /home/student/.ros/log/2e06e492-78b9-11e9-b0bd-000c293d2fe3/follower-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor... Hi !
... shutting down processing monitor complete
done
  
8. Please, describe detailedly what difficulty you are in 

Hi !
Sorry for the inconvenience. I'm looking to execute the following demo, I've done all the steps including installing sklearn. After some research I have the impression that this module''sklearn.ensemble.forest'' no longer exists so I don't know what to do...
Thank you in advance for your help!

RyanJaehyunShim, 2019-05-17:
@ColinJou 

Reinstall python and see if it works. When you install python after removing it, refer to the link below.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#anaconda)

Regards,
Ryan

ColinJou, 2019-05-17:
That's effectively strange because it seem's that I have ''python2.7'', ''python3'' and ''python3.5'' installed, sklearn seems to be installed in the python3 while the terminal said that it is in the 3.5 and to execute the ROS and turtlebot thing I use the 2.7...

RyanJaehyunShim, 2019-05-20:
@ColinJou 

Remove all of them and only reinstall python 2.7 following the link I sent to you. 

Let us know if it doesn't work, 
Ryan

ColinJou, 2019-05-23:
Ok, I started the whole ''Remote PC'' part over again on a new clean virtual machine and everything works fine, thank you very much!

closed 2019-05-23 by RyanJaehyunShim