ROBOTIS-GIT/turtlebot3 #428: 'ascii' codec can't decode byte 0xe4 in position 0: ordinal not in range(128)

Babaevzhu, 2019-05-20:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [*] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [*] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [*] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [*] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.1.0]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    -[Turtlebot3] roslaunch turtlebot3_bringup turtlebot3_robot.launch -- screen
    -[Remote PC] roslaunch turtlebot3_slam turtlebot3_slam.launch
 
7. Copy and Paste your error message on terminal

    - 'ascii' codec can't decode byte 0xe4 in position 0: ordinal not in range(128)
  
8. Please, describe detailedly what difficulty you are in 

    - I have connect turtlebot3 and my PC, since the turtlebot3 can move based on  "roslaunch turtlebot3 _ teleop turtlebot3 _ teleop _ key . launch" command in my remote PC. However, when I run the slam node "roslaunch turtlebot3_slam turtlebot3_slam.launch" in my remote PC, it report an error " 'ascii' codec can't decode byte 0xe4 in position 0: ordinal not in range(128)". I have no idea how to solve this issue, since I have tried to find the solution on the internet, some people said it's a python error, but this node is written by cpp. Hoping for your reply, thank u very much.

RyanJaehyunShim, 2019-05-20:
@Babaevzhu 

We tested it and confirmed it works well. First, remove python and test it again as SLAM does not need python. If it doesn't work, since there is no reason we can think of for the issue, we suggest you reinstall OS and all packages and test it again.

Regards,
Ryan

Babaevzhu, 2019-05-22:
> @Babaevzhu
> 
> We tested it and confirmed it works well. First, remove python and test it again as SLAM does not need python. If it doesn't work, since there is no reason we can think of for the issue, we suggest you reinstall OS and all packages and test it again.
> 
> Regards,
> Ryan

Thanks you very much, I recompile the code, and the problem has been solved.
Regards

closed 2019-05-22 by RyanJaehyunShim