ROBOTIS-GIT/turtlebot3 #429: Sharp IR sensor with Turtlebot3

MHammadA, 2019-05-21:
Hi everyone,

I am working with **Turtlebot3 waffle pi**. And i wanted to use **Sharp IR sensors** for my project. But i wanted to know that is it possible to connect **Sharp IR sensor** with **Raspberry pi**? and also i wanted to code for IR sensor in my existing program, so is their any **ROS** coding example for IR sensor? I am not very expert in ROS and i am new with raspberry pi so please any help will be much appreciated.

Thanks in advance

RyanJaehyunShim, 2019-05-21:
@MHammadA 

As your question is related to ROS & Raspberry pi rather than TB3, please ask the question at the link below. 
(https://answers.ros.org/questions/)

Regards,
Ryan

closed 2019-05-21 by RyanJaehyunShim

MHammadA, 2019-05-21:
Hi Ryan

Thanks for the reply. My question was also that is it possible to connect IR sensor with Raspberry pi. I followed turtlebot emanual and in that manual different sensors are used to connect with opencr but my question was about connecting sensor with raspberry pi. I think this is related to turtlebot3. please help if you know something about that.

Thanks.

RyanJaehyunShim, 2019-05-21:
@MHammadA 

The current tb3 code will not work that way so you will need to modify the source code.

If your question is about the feasibility of connecting IR sensor to Raspberry Pi and using it, please ask the question at the link below.
(https://www.raspberrypi.org/forums/)

Regards,
Ryan