ROBOTIS-GIT/turtlebot3 #430: Mapping

Babaevzhu, 2019-05-22:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [* ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [* ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [* ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [* ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - roslaunch turtlebot3_slam turtlebot3_slam.launch
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - I can running the slam node, but the mapping result is poor. When the turtlebot run in a straight line, the map is good as expected. But when the turtlebot is turning, the map will confuse and be built in wrong places. Is this problem caused by the odometry error? Do I need to calibrate the odomtery? Thank you very much.

RyanJaehyunShim, 2019-05-22:
@Babaevzhu 

If the speed of the robot is too fast, that could happen. Would you check if the robot odom values are correct while turning using $ rostopic echo [topic_name] 

If you still need more help with this issue, attach screenshots of your Rviz after mapping that can help us figure out the problem better.

Regards,
Ryan

Babaevzhu, 2019-05-23:
Thanks for your replying. The turning speed is just 0.1, it should not be too fast. 
The odom output look regular as following, and I upload an image of mapping result. Thanks very much.
------------------------
header: 
  seq: 53
  stamp: 
    secs: 1455209666
    nsecs: 284486028
  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: 0.00025000385358
      y: 0.000497851578984
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: -0.33943733573
      w: 0.940628647804
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: -0.000650920555927
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 54
  stamp: 
    secs: 1455209670
    nsecs: 889877048
[image]

  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: 0.000250007113209
      y: 0.000497855537105
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: -0.3393920362
      w: 0.940645039082
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 4.78752144772e-06
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 55
  stamp: 
    secs: 1455209671
    nsecs: 381521108
  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: 0.000250007113209
      y: 0.000497855537105
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: -0.339320152998
      w: 0.940670967102
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.000505846284796
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 56
  stamp: 
    secs: 1455209674
    nsecs: 542609968
  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: 0.000250008597504
      y: 0.000497857341543
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: -0.339368492365
      w: 0.940653502941
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.000126221595565
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

RyanJaehyunShim, 2019-05-23:
@Babaevzhu 

What do you mean by "The odom output look regular as following" ? 

Regards,
Ryan

Babaevzhu, 2019-05-23:
Sorry, I mean I put the "rostopic echo odom" output at the end of comment.

I face another issue now, when roslaunch the slam node, the map can't be built and always reports a warning "Failed to compute laser pose, aborting initialization (Lookup would require extrapolation into the future.  Requested time 1558579608.604781301 but the latest data is at time 1558579606.448156116, when looking up transform from frame [odom] to frame [base_footprint])"
DO you know how to solve this issue?

Best Regards

RyanJaehyunShim, 2019-05-24:
@Babaevzhu 

Instead of just copypasting the terminal output, you should check it with the actual robot's movement. And If you think there is no problem with odometry, try below and see if it works.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc)

If it does not work, reinstall Ubuntu & ROS packages following the tutorial. You should let us know if you make any modifications while following it. 
(http://emanual.robotis.com/docs/en/platform/turtlebot3/getting_started/#getting-started)

Let us know if it doesn't work,
Ryan

ROBOTIS-Kayman, 2019-07-19:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-07-19 by ROBOTIS-Kayman