ROBOTIS-GIT/turtlebot3 #431: TB3-Ros2 no response after Ubuntu 18.04.2 upgrade to latest patch.

rafaelhuang-intel, 2019-05-27:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - Waffle & Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Intel NUC (I7-6700)

3. Which OS you installed in SBC?

    - Ubuntu 18.04 LTS (Bionic Beaver)

Below is execution steps:
1. install Ubuntu 18.04.2 LTS (Bionic Beaver) (kernal 4.15.x or 4.18.x)
2. sudo apt update
3. sudo apt upgrade
4. after all ubuntu patch updated, install all TB3 relate package, like TB3, Navigation2, Cartographer, etc.
5. run TB3 and Cartographer SLAM (or navigation)

All data show on terminal:

intel@intel-desktop:~/turtlebot3$ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit

cd ~/turtlebot3 && MicroXRCEAgent udp 2018
UDP agent initialization... OK
Enter 'q' for exit
RTPS Participant matched 1.f.59.93.f4.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.fd.43.0.0.1.0.0.0|0.0.1.c1
RTPS Subscriber matched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.22.4
RTPS Participant matched 1.f.59.93.b.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.15.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.14.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.14.44.0.0.2.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.16.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.15.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.b.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.14.44.0.0.2.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.14.44.0.0.1.0.0.0|0.0.1.c1
RTPS Subscriber unmatched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.22.4
RTPS Participant unmatched 1.f.59.93.fd.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.f4.43.0.0.1.0.0.0|0.0.1.c1

intel@intel-desktop:~$ ~/turtlebot3/turtlebot3_lidar
turtlebot3 lidar client setup complete!
1 - Sent topic: 2.101000, range_max: 361
2 - Sent topic: 2.127000, range_max: 361
3 - Sent topic: 2.128000, range_max: 361
4 - Sent topic: 2.114000, range_max: 361
5 - Sent topic: 2.128000, range_max: 361
6 - Sent topic: 2.144000, range_max: 361
7 - Sent topic: 2.133000, range_max: 361
8 - Sent topic: 2.121000, range_max: 361
9 - Sent topic: 2.133000, range_max: 361
10 - Sent topic: 2.102000, range_max: 361
11 - Sent topic: 2.129000, range_max: 361
12 - Sent topic: 2.091000, range_max: 361
13 - Sent topic: 2.100000, range_max: 355
14 - Sent topic: 2.140000, range_max: 361
15 - Sent topic: 2.116000, range_max: 361
16 - Sent topic: 2.115000, range_max: 361
17 - Sent topic: 2.128000, range_max: 361
18 - Sent topic: 2.123000, range_max: 361
19 - Sent topic: 2.134000, range_max: 361
20 - Sent topic: 2.125000, range_max: 361
[... 83 lines cut ...]
[cartographer_node-1] I0527 01:26:40.000000 17428 configuration_file_resolver.cc:41] Found '/home/intel/robot_sdk/sdk_ws/device_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] I0527 01:26:40.000000 17428 configuration_file_resolver.cc:41] Found '/home/intel/robot_sdk/sdk_ws/device_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[cartographer_node-1] I0527 01:26:40.000000 17428 submap_2d.cc:187] Added submap 1
[cartographer_node-1] I0527 01:26:40.000000 17428 map_builder_bridge.cc:132] Added trajectory with ID '0'.
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[rviz2-3] [WARN] [rviz2]: Could not find folder /opt/robot_sdk/core/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
[rviz2-3] [WARN] [rviz2]: Could not find folder /opt/robot_sdk/core/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty

**BTW, if not upgrade all ubuntu patch (no run step3), this issue disappear.**  Does anyone met the same issue before, would you please share some experience about how to fix or debug?

yoshimalucky, 2019-05-28:
No response of what ...?

rafaelhuang-intel, 2019-06-03:
@yoshimalucky 
intel@intel-desktop:~/turtlebot3$ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit
no data output

cartographer no map show in rviz2, always output "[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty" in terminal.

OpusK, 2019-06-03:
@rafaelhuang-intel 

Currently, MicroXRCEAgent only supports Crystal Patch3 and below. If you use a later version, it may not work (perhaps this will cause an error already in the compilation process!)

Recently Dashing has been updated, Agent and Client are still working on it. (See eProsima's Repository.)
If this work is preceded, ros2arduino which is used in TB3 will be updated, and it will be possible to test later in the latest version.

rafaelhuang-intel, 2019-06-03:
@OpusK currently I used commit (version: 74ab7d57c7821053ccdad486f57d53f46dfc7b88) in my environment for build and test,  if not upgrade all ubuntu patch (no run above step3), this issue disappear, so I think this issue may cause by Ubuntu patch conflict with MicroXRCEAgent.

OpusK, 2019-06-12:
@rafaelhuang-intel 

Thank you for the information.
If this phenomenon is reproduced again, I will forward this issue to eProsima.

rafaelhuang-intel, 2019-06-14:
@OpusK this issue still exist.

intel@intel-desktop:~$ uname -a
Linux intel-desktop 4.18.0-21-generic #22~18.04.1-Ubuntu SMP Thu May 16 15:07:19 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
intel@intel-desktop:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

OpusK, 2019-06-14:
Hi, @rafaelhuang-intel 

The Agent is currently being updated.
Can you tell me which version of ROS2 you are using?
If you need to continue using the Crystal version, would you like to leave this issue on [eProsima](https://github.com/eProsima/Micro-XRCE-DDS-Agent)?

rafaelhuang-intel, 2019-06-14:
Hi, @OpusK 

I tested on ROS2 Dashing release now. I will report this issue to eProsima.

thanks !

OpusK, 2019-06-14:
@rafaelhuang-intel 

To test in Dashing, you must use the correct version of Agent, Client, and TB3 for Dashing.
However, the opencr core f/w for Dashing and ROS2 packages are not released yet.

Just as you have already raised issues with eProsima, please test it with the crystal version (patch3) before release.

OpusK, 2019-06-14:
@rafaelhuang-intel 

Looking at your logs, the lidar client seems to work, and the Agent seems to work fine.
Would you like to check if the `ros2 topic echo / scan_half` command displays the scan_half topic?

rafaelhuang-intel, 2019-06-14:
@OpusK I think lidar work fine, but MicroXRCEAgent serial /dev/ttyACM0 no data. if not upgrade Ubuntu patch, more data can show in terminal.

rafaelhuang-intel, 2019-06-17:
@OpusK As I checked again, the MicroXRCEAgent serial /dev/ttyACM0 still no data show in terminal, 
the   ** ros2 topic echo /scan_half** can get scan data, below are get log:

intel@intel-desktop:~$ ros2 topic echo /scan_half
header:
  stamp:
    sec: 1560738401
    nanosec: 475105849
  frame_id: base_scan
angle_min: 0.0
angle_max: 6.2657318115234375
angle_increment: 0.01745329238474369
time_increment: 2.9880000511184335e-05
scan_time: 0.0
range_min: 0.11999999731779099
range_max: 3.5
ranges: [0.20600000023841858, 0.20399999618530273, 0.21699999272823334, 0.21899999678134918, 0.2160000056028366, 0.21199999749660492, 0.210999995470047, 0.210999995470047, 0.21299999952316284, 0.2160000056028366, 0.22100000083446503, 0.2280000001192093, 0.24199999868869781, 0.0, 0.9049999713897705, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.8489999771118164, 0.0, 2.759000062942505, 0.0, 3.3359999656677246, 3.2190001010894775, 0.5809999704360962, 0.5740000009536743, 0.578000009059906, 0.628000020980835, 2.8929998874664307, 2.8450000286102295, 2.7960000038146973, 2.7279999256134033, 0.0, 1.684999942779541, 3.3519999980926514, 2.23799991607666, 2.359999895095825, 2.9639999866485596, 3.5369999408721924, 3.191999912261963, 2.634000062942505, 0.0, 0.0, 0.0, 3.117000102996826, 2.5199999809265137, 2.881999969482422, 3.2070000171661377, 0.0, 0.0, 3.1630001068115234, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.4189999103546143, 2.48799991607666, 0.0, 0.0, 0.0, 0.0, 0.796999990940094, 0.7350000143051147, 0.8059999942779541, 0.0, 3.5929999351501465, 0.0, 0.0, 0.0, 3.549999952316284, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 4.0980000495910645, 1.8860000371932983, 1.1890000104904175, 1.1069999933242798, 0.7360000014305115, 0.7319999933242798, 0.7229999899864197, 0.7210000157356262, 0.7080000042915344, 0.7080000042915344, 0.7089999914169312, 0.7099999785423279, 0.7160000205039978, 0.718999981880188, 0.7200000286102295, 0.718999981880188, 0.7310000061988831, 0.7409999966621399, 0.7440000176429749, 0.75, 0.7630000114440918, 0.7680000066757202, 0.7839999794960022, 0.796999990940094, 0.7929999828338623, 0.796999990940094, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.4690000116825104, 0.5, 0.5070000290870667, 0.5220000147819519, 0.5379999876022339, 0.0, 0.6100000143051147, 0.6380000114440918, 0.6290000081062317, '...']
intensities: [0.0]

header:
  stamp:
    sec: 1560738401
    nanosec: 678190407
  frame_id: base_scan
angle_min: 0.0
angle_max: 6.2657318115234375
angle_increment: 0.01745329238474369
time_increment: 2.990000029967632e-05
scan_time: 0.0
range_min: 0.11999999731779099
range_max: 3.5
ranges: [0.20600000023841858, 0.20399999618530273, 0.21699999272823334, 0.21899999678134918, 0.2160000056028366, 0.21199999749660492, 0.210999995470047, 0.210999995470047, 0.21299999952316284, 0.2150000035762787, 0.2199999988079071, 0.22699999809265137, 0.23999999463558197, 0.0, 0.949999988079071, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.8329999446868896, 0.0, 2.749000072479248, 0.0, 3.36299991607666, 3.243000030517578, 0.5849999785423279, 0.574999988079071, 0.5799999833106995, 0.593999981880188, 2.9049999713897705, 2.8550000190734863, 2.803999900817871, 2.743000030517578, 0.0, 1.753000020980835, 3.364000082015991, 2.2780001163482666, 2.3440001010894775, 2.9719998836517334, 3.309000015258789, 3.1710000038146973, 2.5910000801086426, 2.6059999465942383, 0.24500000476837158, 2.513000011444092, 3.1410000324249268, 2.5299999713897705, 2.6989998817443848, 3.2190001010894775, 0.0, 0.0, 3.1519999504089355, 0.0, 0.0, 3.140000104904175, 0.0, 0.0, 0.0, 2.4489998817443848, 0.0, 0.0, 0.0, 0.0, 0.0, 0.8109999895095825, 0.7789999842643738, 0.7699999809265137, 0.0, 3.5980000495910645, 0.0, 0.0, 0.0, 3.5490000247955322, 0.0, 0.0, 0.0, 0.0, 0.0, 4.166999816894531, 0.0, 4.015999794006348, 1.878000020980835, 1.1929999589920044, 1.1759999990463257, 0.7379999756813049, 0.7390000224113464, 0.7279999852180481, 0.7129999995231628, 0.7170000076293945, 0.7149999737739563, 0.7110000252723694, 0.7160000205039978, 0.7139999866485596, 0.7129999995231628, 0.7170000076293945, 0.7210000157356262, 0.7369999885559082, 0.7409999966621399, 0.7429999709129333, 0.7480000257492065, 0.7570000290870667, 0.7630000114440918, 0.7739999890327454, 0.7860000133514404, 0.7929999828338623, 0.8059999942779541, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.46799999475479126, 0.5, 0.5059999823570251, 0.5210000276565552, 0.5299999713897705, 0.0, 0.6119999885559082, 0.6389999985694885, 0.6290000081062317, '...']
intensities: [0.0]

header:
  stamp:
    sec: 1560738402
    nanosec: 76724230
  frame_id: base_scan
angle_min: 0.0
angle_max: 6.2657318115234375
angle_increment: 0.01745329238474369
time_increment: 2.990000029967632e-05
scan_time: 0.0
range_min: 0.11999999731779099
range_max: 3.5
ranges: [0.2070000022649765, 0.20499999821186066, 0.20800000429153442, 0.21899999678134918, 0.2160000056028366, 0.21199999749660492, 0.210999995470047, 0.210999995470047, 0.21299999952316284, 0.2150000035762787, 0.2199999988079071, 0.2280000001192093, 0.24199999868869781, 0.0, 0.968999981880188, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.8550000190734863, 0.0, 2.744999885559082, 0.0, 3.3399999141693115, 3.24399995803833, 0.5839999914169312, 0.574999988079071, 0.5809999704360962, 0.5950000286102295, 2.9159998893737793, 2.8410000801086426, 2.7960000038146973, 2.746999979019165, 0.0, 1.7289999723434448, 3.367000102996826, 2.2690000534057617, 2.3469998836517334, 2.9719998836517334, 3.436000108718872, 3.1630001068115234, 2.648000001907 [... cut ...] 995, 0.7210000157356262, 0.7170000076293945, 0.7239999771118164, 0.7409999966621399, 0.7490000128746033, 0.7390000224113464, 0.7480000257492065, 0.7649999856948853, 0.7710000276565552, 0.7730000019073486, 0.7889999747276306, 0.8100000023841858, 0.8029999732971191, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.47999998927116394, 0.49900001287460327, 0.5070000290870667, 0.5210000276565552, 0.5329999923706055, 0.0, 0.609000027179718, 0.6370000243186951, 0.628000020980835, '...']
intensities: [0.0]

OpusK, 2019-06-17:
@rafaelhuang-intel 

When you build Agent using `cmake`, try the options below and try Serial again. (For log check)
```
$ cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=4096-DCONFIG_SERIAL_TRANSPORT_MTU=4096 -DVERBOSE=ON ..
```

rafaelhuang-intel, 2019-06-17:
@OpusK 
I tried again, the Serial still no data output.
and the "MicroXRCEAgent udp 2018" popup mor warning log, 

intel@intel-desktop:~/turtlebot3$ cd ~/turtlebot3 && MicroXRCEAgent udp 2018
UDP agent initialization... OK
Enter 'q' for exit

<== [Client connected | session: 0x81 | client key: 0xCC 0xCC 0xDD 0xDD ]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0011 | PARTICIPANT]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0012 | TOPIC | participant id: 0x0011]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0013 | PUBLISHER | participant id: 0x0011]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0015 | DATA_WRITER | publisher id: 0x0013 | xml: 0x00000098]
RTPS Participant matched 1.f.59.93.7a.2a.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.a1.3d.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.ab.3d.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.aa.3d.0.0.1.0.0.0|0.0.1.c1
RTPS Subscriber matched 1.f.59.93.ab.3d.0.0.1.0.0.0|0.0.22.4
RTPS Participant matched 1.f.59.93.85.50.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.f7.50.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.f7.50.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.f6.50.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.f6.50.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.f6.50.0.0.2.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.f8.50.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.f8.50.0.0.1.0.0.0|0.0.1.c1

OpusK, 2019-06-17:
@rafaelhuang-intel 

> <== [Client connected | session: 0x81 | client key: 0xCC 0xCC 0xDD 0xDD ]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0011 | PARTICIPANT]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0012 | TOPIC | participant id: 0x0011]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0013 | PUBLISHER | participant id: 0x0011]
<== [Create XRCE object | client key: 0xCC 0xCC 0xDD 0xDD | id: 0x0015 | DATA_WRITER | publisher id: 0x0013 | xml: 0x00000098]

The above log should also appear in the Serial Agent.
Can you tell me the order in which you ran the Serial Agent?

To explain again from my side, connect the OpenCR and check the OpenCR's dev name (eg. /dev/ttyACM0) before running the Serial Agent. (Including baud rate command)

For example,
```bash
# After reset OpenCR (Because of the nature of the Serial Port, this order is very important.)
$ ls /dev    # for check OpenCR dev name.
$ MicroXRCEAgent serial /dev/ttyACM0 1000000
```

rafaelhuang-intel, 2019-06-17:
@OpusK 

```
Termina 1
$intel@intel-desktop:~/turtlebot3$ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
Termina 2
$intel@intel-desktop:$ cd ~/turtlebot3 && MicroXRCEAgent udp 2018
Termina 3
$intel@intel-desktop:~$ ~/turtlebot3/turtlebot3_lidar
Termina 4
$ ros2 launch turtlebot3_bringup robot.launch.py
```

all terminal have more log output expect terminal 1

rafaelhuang-intel, 2019-06-17:
@OpusK 

after run $ ls /dev    # for check OpenCR dev name , no data output. so it's means opencr serial port not connect?

OpusK, 2019-06-17:
@rafaelhuang-intel 

Is there any possibility of connecting other devices besides OpenCR?
In addition, reset OpenCR and use the normal terminal program (eg screen) to see what data comes from OpenCR?

rafaelhuang-intel, 2019-06-17:
@OpusK 

Sorry, I tried another workable PC(not upgrade Ubuntu 18.04 all patch), the log can show in terminal, it's my mistake. 

so could you share your recommendation which device can double check besides OpenCR?

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

This issue will be solved to download new firmware

closed 2019-08-21 by routiful