ROBOTIS-GIT/turtlebot3 #433: Turtlebot 3 will not move to nav point. Says it has reached point but moves only a bit or not at all.

extrasleepy, 2019-05-30:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ X ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ X ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ X ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ X ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - roscore
     - roslaunch turtlebot3_bringup turtlebot3_robot.launch (on pi)
     roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml
 
7. Copy and Paste your error message on terminal

    - terminal says navigation point reached but robot is not at navgation point. There is no error in terminal.
  
8. Please, describe detailedly what difficulty you are in 

      - terminal says navigation point reached but robot is not at navgation point. There is no error in terminal. Other robots in the classroom, with same setup are not having this problem.

yoshimalucky, 2019-06-03:
Can you copy and paste the EXACTLY original terminal message? (Do not modify in your sentence.)

I googled "navigation point reached but robot is not at navigation point" in your sentence, but this issue came to the top result. This seems to be a common warning or error message related with the navigation package, so we should find some resources if we googled the exactly same terminal message.

ROBOTIS-Kayman, 2019-06-03:
Hi,

Could you tell me more details about your problem?
(ex. terminal screen shot, opencr firmware version, etc..)

Regards, 
Kayman

extrasleepy, 2019-06-03:
Hello,

Thank you for your help.

The firmware is v1.2.2

I’ve attached some of the terminal output:

unsalted, 2019-06-03:
Running into the exact same issue. I have reflashed the pi completely and updated all software.

unsalted, 2019-06-03:
Mine clears exits on: [ERROR] [1559581154.245286582]: Aborting because the robot appears to be oscillating over and over. Even after executing all recovery behaviors

ROBOTIS-Kayman, 2019-06-04:
@extrasleepy,
Could you attach the log again?
I can not see it.

Regards, 
kayman

ROBOTIS-Kayman, 2019-06-04:
@unsalted 
Can I see the whole log of terminal and rviz screen shot?
It seems that you ran TB3 in the narrow environments. so the navigation package did not seem to find the path.

Regards,
Kayman

extrasleepy, 2019-06-04:
I understand that this environment looks small but I have also tried wide open environments as well. Even with nothing around, the turtlebot will not move to a nav point. If I make a nav point that is very far away, the turtlebot will move slightly then stop. It never moves more than a few centimeters but the terminal output always says “Got new plan” then “Goal reached". The path never turns red like it does on some of the other turtlebots I’ve used. I’ll send you the rviz and output tomorrow. I don’t have the bot tonight. I’ll use a wide open environment, though that has not helped in the past. Thanks again.

extrasleepy, 2019-06-05:
Here's the terminal output:
roslaunch turtlebot3_navigation turtlebot3_navigation.launch
map_file:=$HOME/map.yaml
... logging to
/home/electronics-03/.ros/log/7a59b3d6-8725-11e9-889d-080027427939/roslaunch-electronics03-VirtualBox-30823.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.1.8.119:36125/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/gui_publish_rate: 50.0
 * /amcl/initial_pose_a: 0.0
 * /amcl/initial_pose_x: 0.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.02
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 180
 * /amcl/laser_max_range: 3.5
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 3000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.1
 * /amcl/odom_alpha2: 0.1
 * /amcl/odom_alpha3: 0.1
 * /amcl/odom_alpha4: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 0.5
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.2
 * /move_base/DWAPlannerROS/acc_lim_theta: 3.2
 * /move_base/DWAPlannerROS/acc_lim_x: 2.5
 * /move_base/DWAPlannerROS/acc_lim_y: 0.0
 * /move_base/DWAPlannerROS/controller_frequency: 5.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.325
 * /move_base/DWAPlannerROS/goal_distance_bias: 20.0
 * /move_base/DWAPlannerROS/latch_xy_goal_tolerance: False
 * /move_base/DWAPlannerROS/max_rot_vel: 2.75
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_trans_vel: 0.22
 * /move_base/DWAPlannerROS/max_vel_x: 0.22
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_rot_vel: 1.37
 * /move_base/DWAPlannerROS/min_trans_vel: 0.11
 * /move_base/DWAPlannerROS/min_vel_x: -0.22
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
 * /move_base/DWAPlannerROS/occdist_scale: 0.02
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/DWAPlannerROS/path_distance_bias: 32.0
 * /move_base/DWAPlannerROS/publish_cost_grid_pc: True
 * /move_base/DWAPlannerROS/publish_traj_pc: True
 * /move_base/DWAPlannerROS/scaling_speed: 0.25
 * /move_base/DWAPlannerROS/sim_time: 1.2
 * /move_base/DWAPlannerROS/stop_time_buffer: 0.2
 * /move_base/DWAPlannerROS/vth_samples: 40
[... 95 lines cut ...]
[ INFO] [1559693258.604741819]: Got new plan
[ INFO] [1559693258.804779118]: Got new plan
[ INFO] [1559693259.005052972]: Got new plan
[ INFO] [1559693259.204688254]: Got new plan
[ INFO] [1559693259.404815628]: Got new plan
[ INFO] [1559693259.604950299]: Got new plan
[ INFO] [1559693259.805100026]: Got new plan
[ INFO] [1559693260.005268695]: Got new plan
[ INFO] [1559693260.204704891]: Got new plan
[ INFO] [1559693260.405101013]: Got new plan
[ INFO] [1559693260.604703816]: Got new plan
[ INFO] [1559693260.804856644]: Got new plan
[ INFO] [1559693261.005976894]: Got new plan
[ INFO] [1559693261.205093157]: Got new plan
[ INFO] [1559693261.405094149]: Got new plan
[ INFO] [1559693261.604697219]: Got new plan
[ INFO] [1559693261.804833335]: Got new plan
[ INFO] [1559693262.004799068]: Got new plan
[ INFO] [1559693262.205570928]: Got new plan
[ INFO] [1559693262.405527621]: Got new plan
[ INFO] [1559693262.605282632]: Got new plan
[ INFO] [1559693262.605434131]: Goal reached

here's the rviz in which the robot is not making it to the nav point:
[image: image.png]

extrasleepy, 2019-06-10:
[image]

ROBOTIS-Kayman, 2019-06-11:
It's hard to figure it out what the problem is.
Could you try other example like [tele-op](http://emanual.robotis.com/docs/en/platform/turtlebot3/teleoperation/#keyboard)?

I suspect odom topic from OPENCR.
Would you like to test the navigation after changing OPENCR to something else?

extrasleepy, 2019-06-11:
The tele-op works well. I do think it has something to do with the OPENCR. I reinstalled all the software on the Pi and on the host computer and still the problem exists. I think the OPENCR odom must have gotten damaged. Do you think that sounds right? thank you.

ROBOTIS-Kayman, 2019-06-11:
It makes sense.
Would you like to test the navigation after changing OPENCR to something else, if possible?

extrasleepy, 2019-06-11:
Yes. I'll try that. Thank you for your help.

closed 2019-06-11 by extrasleepy