ROBOTIS-GIT/turtlebot3 #435: Problem with turtlebot3_example precisely with turtlebot3_client and turtlebot3_server

Hosbye, 2019-06-04:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3

3. Which OS you installed in SBC?

    - [ ] Raspbian

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - HERE

Hosbye, 2019-06-04:
Hello, I'm new to ROS and I have a problem to test the turtlebot3_example in order to move to custom route (square). When I tested the code according the following tutorial, everything was Ok, but when I put my value (s 0.5 0), Nothing was happening (my robot hasn't moved and the screen was as frozen) : https://youtu.be/Xg1pKFQY5p4
Could you help me please ?

PS: Be forgiving for my English, I am not English but I try to write well :-)

RyanJaehyunShim, 2019-06-05:
@Hosbye 

We have tested that example and confirmed it works fine. 

1. Did you encounter any problems when you were doing tutorial 1~3? 
    (Your questions is about tutorial 4)
2. Would you send us screenshots of your terminal windows so that we can see if you are following the tutorial correctly? 

Regards,
Ryan

Hosbye, 2019-06-05:
@RyanJaehyunShim 
Thank you for your help, My steps were the following
[image]
[image]
[image]
[image]
[image]

Best regards,
Hosbye

kijongGil, 2019-06-11:
Hi, @Hosbye :)
Could you command export ROS_MASTER_URI before running the turtlebot3_server node in terminal?
And why you don't insert ROS_MASTER_URI in your .bashrc?

Thanks,
Gilbert

ROBOTIS-Kayman, 2019-06-11:
Hi,

I think your network setting of ROS is the problem.
Would you like to follow [this e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#network-configuration), and try it again?

Regards,
Kayman

Hosbye, 2019-06-11:
Hi, @kijongGil :)
Yes, I can.
And of course, I have already inserted "ROS_MASTER_URI in my .bashrc". But when I run the roscore and the other commands, it didn't work before.

I tried to command "export ROS_MASTER_URI" before running the turtlebot3_server node in terminal, but it doesn't work like in the previous commands.

I retry inserted "ROS_MASTER_URI in my .bashrc" and re-source it. And I've tested all steps without  command export ROS_MASTER_URI (except in "pi"), but it doesn't work like in the previous commands :'(

Thanks Gilbert,
Hosbye

kijongGil, 2019-06-11:
@Hosbye You mean that when you run roscore without "export ROS_MASTER_URI" in terminal, it didn't work? If yes, you have a problem about installation ROS. Could you reinstall ROS and try again?

Hosbye, 2019-06-11:
@kijongGil I've reinstalled ROS, inserted "ROS_MASTER_URI in my .bashrc" and re-source it, and now the roscore work without "export ROS_MASTER_URI" in terminal, except in the TurtleBot, when the command " roslaunch turtlebot3_bringup turtlebot3_robot.launch" doesn't work without export.

And the turtlebot3_server node and turtlebot3_client node doesn't work like in the previous commands :'(

Hosbye, 2019-06-11:
Does it mean that I've to reinstall ROS in my TurtleBot?

RyanJaehyunShim, 2019-06-11:
@Hosbye 

Would you read the following link again carefully and test it again? 
(http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup)

You should do set up both your laptop and the turtlebot. 

Regards,
Ryan

Hosbye, 2019-06-12:
Hello, @RyanJaehyunShim 
I've the problem with "export ROS_MASTER_URI". But the turtlebot3_server node and turtlebot3_client node doesn't work.

Hosbye, 2019-06-12:
Hi, @kijongGil 
Could you please send, me, your original turtlebot3_server node and turtlebot3_client node? There is some difference between the program download with ROS and this one : https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_example/nodes/turtlebot3_server

Thanks,
Hosbye

kijongGil, 2019-07-05:
Hi, @Hosbye 
I'm sorry for late reply. I don't have a my original code. It's same in github. 
You can see this line https://github.com/ROBOTIS-GIT/turtlebot3/blob/d3cdcc6647812ae9a83f05e626cdae322923ac84/turtlebot3_example/nodes/turtlebot3_client#L74) and set the time out. (ex, client.wait_for_result(1000))
If you get the time out error, server node has problem.

ROBOTIS-Kayman, 2019-08-21:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-08-21 by ROBOTIS-Kayman