ROBOTIS-GIT/turtlebot3 #438: Error running roslaunch turtlebot3_bringup turtlebot3_robot.launch on remote PC

nonso1234, 2019-06-13:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    -  Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    -  Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    -  Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    Firmware version: [x.x.x]
 
6. Write down the commands you used in order

     roslaunch turtlebot3_bringup turtlebot3_robot.launch 
 
7. Copy and Paste your error message on terminal

    started roslaunch server http://192.168.1.118:45767/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.1.118:11311

process[turtlebot3_core-1]: started with pid [4455]
process[turtlebot3_lds-2]: started with pid [4456]
process[turtlebot3_diagnostics-3]: started with pid [4457]
**[ERROR] [1560387263.586563441]: An exception was thrown: open: No such file or directory**
[INFO] [1560387263.789426]: ROS Serial Python Node
[INFO] [1560387263.795352]: Connecting to /dev/ttyACM0 at 115200 baud
**[ERROR] [1560387263.797172]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0:** **[Errno 2] No such file or directory: '/dev/ttyACM0'**
[turtlebot3_lds-2] process has died [pid 4456, exit code 255, cmd /home/chinonso/catkin_ws/devel/lib/hls_lfcd_lds_driver/hlds_laser_publisher __name:=turtlebot3_lds __log:=/home/chinonso/.ros/log/40fa7676-8d69-11e9-98cb-1866da2dc802/turtlebot3_lds-2.log].
log file: /home/chinonso/.ros/log/40fa7676-8d69-11e9-98cb-1866da2dc802/turtlebot3_lds-2*.log
[turtlebot3_core-1] process has finished cleanly
log file: /home/chinonso/.ros/log/40fa7676-8d69-11e9-98cb-1866da2dc802/turtlebot3_core-1*.log
  
8. Please, describe detailedly what difficulty you are in 

 I am following the turtlebot setup guide. Currently on the SLAM/ Bringup exercise (http://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#slam). Just to confirm, do I run "roslaunch turtlebot3_bringup turtlebot3_robot.launch" on my remote PC or turtlebot PC. It says **[Turtlebot]** on the header. I got the error(highlighted on 7)  running it on my remote PC. If I'm doing this correctly, then I would like to know what I'm doing wrong. Thanks a lot

Pallav1299, 2019-06-16:
You need to roslaunch the "**turtlebot3_robot.launch**" launch file in the "turtlebot3_bringup" on the Turtlebot SBC(Single Board Computer). 
You can use **SSH** to connect to the Turtlebot(preferred) or do a separate hardware setup(including display, keyboard and mouse) and then roslaunch the "turtlebot3_robot.launch" launch file.
Rest of the files are to be run on your remote master computer.

ROBOTIS-Kayman, 2019-06-17:
Hi,

**[Turtlebot]** means that you have to run the launch file in the Turtlebot via ssh.
Could you try again as @Pallav1299 explained?

Regards,
Kayman

nonso1234, 2019-06-20:
It works. Thanks a lot @Pallav1299  @ROBOTIS-Kayman

closed 2019-06-21 by ROBOTIS-Kayman