ROBOTIS-GIT/turtlebot3 #439: Raspberry Pi camera not working

silvara, 2019-06-21:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?
    - [v] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [v] Raspberry Pi 3 B+

3. Which OS you installed in SBC?
    - [v] Raspbian

4. Which OS you installed in Remote PC?
    - [v] Ubuntu 16.04 LTS (Xenial Xerus)

5. Write down software version and firmware version
    - Software version: [1.1.0]
    - Firmware version: [2.0.1]
 
6. Write down the commands you used in order
    - [RemotePC] $ roscore
    - [TurtleBot3] $ export TURTLEBOT3_MODEL=waffle_pi && ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=om_with_tb3 set_lidar_frame_id:=om_with_tb3/base_scan
    - [TurtleBot3] $ ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch
    - [RemotePC] $ ROS_NAMESPACE=om_with_tb3 roslaunch open_manipulator_with_tb3_tools om_with_tb3_robot.launch
    - [RemotePC] $ export TURTLEBOT3_MODEL=waffle_pi && roslaunch open_manipulator_with_tb3_tools slam.launch use_platform:=true
 
7. Copy and Paste your error message on terminal
    - Displays - Camera - Status - Image -> No Image received
    - No camera image shown in Camera panel
  
8. Please, describe detailedly what difficulty you are in 
    - I installed camera package with https://github.com/ROBOTIS-GIT/turtlebot3/issues/397 in raspbian. I can see video with rqt_image_view, but I cannot see any camera image with TurtleBot3 e-Manual 12.5.

RyanJaehyunShim, 2019-06-21:
@silvara 

1. We would like to ask if you are using your own Raspbian Pi board or the one that comes with the robot when you buy it.
2. We suggest you use MATE for the tutorial as it has been tested it works fine on MATE. 
Below is about how to install MATE.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-ubuntu-mate)

We do not fully understand your situation so give me more information (e.g., any changes you made while following the tutorial)

Regards,
Ryan

silvara, 2019-06-24:
1. I used the Raspberry Pi included in the TurtleBot3 package.
2. According to the TurtleBot3 e-Manual, Ubuntu MATE is not support the Raspberry Pi 3 B+.

I just installed ROBOTIS Raspbian from
http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-based-on-raspbian
And I installed OpenCR firmware from
http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation
And then I sent commands that I wrote before, but the camera shows nothing.

ROBOTIS-Kayman, 2019-06-24:
1. Could you type like below to check the topic of camera.
 `$ rostopic echo /om_with_tb3/raspicam_node/image/compressed`
    If it exists, it means that camera works well. 

2. I suspect it is a TF problem. 
**[RemotePC] $ ROS_NAMESPACE=om_with_tb3 roslaunch open_manipulator_with_tb3_tools om_with_tb3_robot.launch**  
Before upper command, you have to export the TB3 MODEL for using a robot model.
Wasn't there error messages?

3. Unofficial Ubuntu MATE 16.04 exists for Pi B+.
This is a [link](https://ubuntu-mate.community/t/ubuntu-mate-for-raspberry-pi-3-model-b/15867/8)

silvara, 2019-06-24:
1. It works well.
2. Sorry for my mistake. I already added export TURTLEBOT3_MODEL=waffle_pi to my ~/.bashrc.

And there is missing command I used. The full command list is like below
```
[RemotePC & TurtleBot3] $ export TURTLEBOT3_MODEL=waffle_pi
[RemotePC] $ roscore
**[RemotePC] $ roslaunch turtlebot3_bringup turtlebot3_remote.launch**
[TurtleBot3] $ ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=om_with_tb3 set_lidar_frame_id:=om_with_tb3/base_scan
[TurtleBot3] $ ROS_NAMESPACE=om_with_tb3 roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch
[RemotePC] $ ROS_NAMESPACE=om_with_tb3 roslaunch open_manipulator_with_tb3_tools om_with_tb3_robot.launch
[RemotePC] $ roslaunch open_manipulator_with_tb3_tools slam.launch use_platform:=true
```

And the RViz throws Warnings
- TF - Status: Warn
- TF - base_footprint - No transform from [base_footprint] to frame [map]
- TF - base_link - No transform from [base_link] to frame [map]
- TF - base_scan - No transform from [base_scan] to frame [map]
- TF - camera_link - No transform from [camera_link] to frame [map]
- TF - camera_rgb_frame - No transform from [camera_rgb_frame] to frame [map]
- TF - camera_rgb_optical_frame - No transform from [camera_rgb_optical_frame] to frame [map]
- TF - caster_back_left_link - No transform from [caster_back_left_link] to frame [map]
- TF - caster_back_right_link - No transform from [caster_back_right_link] to frame [map]
- TF - imu_link - No transform from [imu_link] to frame [map]
- Camera - Status: Warn
- Camera - Image - No Image received

ROBOTIS-Kayman, 2019-06-26:
1. rpi camera  
    - change a launch file(turtlebot3_bringup/launch/turtlebot3_rpicamera.launch) in TB3 like below.  
    ```
    <launch>
      <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
        <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml"/>
        <param name="width" value="640"/>
        <param name="height" value="480"/>
        <param name="framerate" value="15"/>
        <param name="camera_frame_id" value="camera"/>
      </node>
    </launch>
    ```
    

     - change topic name to `/om_with_tb3/raspicam_node/image` and Transport Hint to `compressed` in the Displays of RVIZ.

2. TF
    In this case, you don't have to run the `turtlebot3_remote.launch`.
    ```
    [RemotePC] $ roslaunch turtlebot3_bringup turtlebot3_remote.launch
    ```
    This is a duplicated function of  `[RemotePC] $ ROS_NAMESPACE=om_with_tb3 roslaunch open_manipulator_with_tb3_tools om_with_tb3_robot.launch`

silvara, 2019-06-27:
Answer2 fixed my TF warnings.
But there are still camera issues.

The launch file is completely same with your code. And I fixed my RViz settings(```/om_with_tb3/raspicam_node/image```, ```compressed```) and save & restarted. But the camera window is still black!
Here is my current terminal, rqt, rviz screenshots for your understanding of my situation.
https://imgur.com/a/bHYoDO4

ROBOTIS-Kayman, 2019-06-27:
It's my mistake.

- change a launch file(turtlebot3_bringup/launch/turtlebot3_rpicamera.launch) in TB3 like below.
    ```
    <launch>
      <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
        <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml"/>
        <param name="width" value="640"/>
        <param name="height" value="480"/>
        <param name="framerate" value="15"/>
        <param name="camera_frame_id" value="camera_link"/>
      </node>
    </launch>
    ```

silvara, 2019-06-27:
I changed value of ```camera_frame_id``` to ```camera_rgb_optical_frame```
But there are still no camera images in rviz.
Here is my changed terminal, rqt, rviz screenshots.
https://imgur.com/a/sFqggnO

ROBOTIS-Kayman, 2019-06-27:
It's the TF prefix problem.

- 
    change a launch file(turtlebot3_bringup/launch/turtlebot3_rpicamera.launch) in TB3 like below.
```
    <launch>
      <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
        <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml"/>
        <param name="width" value="640"/>
        <param name="height" value="480"/>
        <param name="framerate" value="15"/>
        <param name="camera_frame_id" value="om_with_tb3/camera_rgb_optical_frame"/>
      </node>
    </launch>
```

silvara, 2019-06-27:
Oh!! it finally fixed.
Thank you for kind reply!

closed 2019-06-27 by silvara