ROBOTIS-GIT/turtlebot3 #440: Cannot see data from LDS

hessex906, 2019-06-21:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    After installing and setting up everything as described on the Turtlebot3 tutorials, we have been unable to get data from the LDS. The LDS rotates, but we cannot see the point cloud in RViz nor can we see anything coming from the scan topic. Echoing scan results in no output and leaves the terminal hanging. It's the exact same problem as described here (https://github.com/ROBOTIS-GIT/turtlebot3/issues/303), however, no solution was found and the issue was closed.

RyanJaehyunShim, 2019-06-27:
@hessex906 

I will get back to you after I test it.

Regards,
Ryan

RyanJaehyunShim, 2019-06-27:
@hessex906 

I just tested and checked it works fine. 
Did you try as the instruction from Gilbert? (https://github.com/ROBOTIS-GIT/turtlebot3/issues/303#issuecomment-418597082)

Let me know if it still doesn't work even after tying that. 

Regards,
Ryan

hessex906, 2019-06-27:
We followed those exact instructions but when we do the echo, the command just hangs and there is no output.

RyanJaehyunShim, 2019-06-28:
@hessex906 

As the software has been tested and there is no problem with it, I suspect the LDS on your robot is broken or you made setup mistakes. 

First, check if the cable connection between the LDS and the robot is done as instruction on the eManual, reinstall all OS and ROS packages on your laptop and robot, and if it still does not work even after that, I recommend you replace it with a new one and test it again.

Regards,
Ryan

ROBOTIS-Kayman, 2019-08-21:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-08-21 by ROBOTIS-Kayman