ROBOTIS-GIT/turtlebot3 #441: Error using gmapping

gutilacerda, 2019-06-24:
I'm trying to open my turtlebot3 burguer in RVIZ with gmapping but I found this:
[image]
[image]

nonso1234, 2019-06-25:
Try expanding global options and select "Odometry" to generate laser scan. That should probably fix it

ROBOTIS-Kayman, 2019-06-26:
It is assumed to be a TF problem.
But it's hard to figure out what the problem is exactly.
Please attach the executed command or terminal log.

Regards,
Kayman

ROBOTIS-Kayman, 2019-07-19:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-07-19 by ROBOTIS-Kayman

Adel-kn, 2019-09-06:
Hello,
I think the problem is that you have no internet access in your router. Try to connect your router to the internet and the problem will be solved