ROBOTIS-GIT/turtlebot3 #442: ROS2 on Turtlebot3

ryanewel, 2019-06-24:
Hi,
Are there any plans to provide a tutorial for running ROS2 directly on a Turtlebot3? Currently the tutorial has you running MicroXRCE DDS agents on the turtlebot and all actual ROS2 nodes are run on a remote machine. It'd nice to be able to run everything as ROS2 nodes on the turtlebot.

Also, are there any plans for the Turtlebot3 packages be released in apt so that they can installed via rosdep instead of needing to install from source?

routiful, 2019-06-25:
We have plan to release it next month.

https://github.com/ROBOTIS-GIT/turtlebot3/issues/426

mkhansen-intel, 2019-07-30:
@routiful - do you have any update on the Dashing release?

OpusK, 2019-07-30:
Hi, @mkhansen-intel 

Currently, most people are on vacation in Korea.

I have made an OpenCR firmware update, and have made an e-manual update.
I did not completely check everything I had to check, but I did update after the simple test.
Please refer to the [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2).

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

closed 2019-08-21 by routiful

ROBOTIS-Will, 2021-04-06:
Sorry about the confusion.
The page has been modified to support more ROS versions.
Please go to below page and select the ROS version at the top of the page.
https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start
Thank you.